FPGAwars/apio #430: Create an Gowin apio project example in test-examples #429

zapta, 2024-09-29:
This issue is for adding to  the``test-examples`` directory a test example for an Gowin board.  Currently the testing of apio for Gowin is limited.

https://github.com/FPGAwars/apio/tree/develop/test-examples

Requirements:
1. The project should have directory structure similar to the other boards their
2. The project should have at least one testbench (see Alhambra II there).

Nice to have
3. The project should be uploadable to the board and functional
4. The project should be an IceStudio project.

zapta, 2024-09-29:
Hi @wireboy5, can you look into this?  Currently we don't have a way to test the Gowin functionality.

cavearr, 2024-11-12:
Hi @zapta i'm try today i'm working on this for icestudio and your work is aligned with mine , in the day i'll give you feedback

zapta, 2024-12-29:
Added a gowin example.  (Not tested yet with an actual board)
https://github.com/FPGAwars/apio-examples/tree/master/examples/sipeed-tang-nano-4k/blinky

closing

closed 2024-12-29 by zapta