FPGAwars/apio #447: Does apio needs to the ftdi_eeprom utility?

zapta, 2024-10-27:
@Obijuan, @cavearr, a question for you.  The build script of the tools-oss-cad-suite package uses the upstream packge  ``tools-system`` and as far as I can see, extracts from it only the utility ``ftdi_eeprom``, for windows and linux (see links below).

My questions for you:
1. Is it correct or is this package is used also for other things in tools-oss-cad-suite?
2. Can it be removed from ``tools-oss-cad-suite``?  (Rationale is that ftdi_eeprom is  needed for board assembly rather than for uploading designs to boards).

Reference links

[TOOL_SYSTEM_URL_BASE=https://github.com/FPGAwars/tools-system/releases/download/v$TOOL_SYSTEM_VERSION
](https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/build.sh#L265)

https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/scripts/install_linux_x64.sh#L13

https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/scripts/install_windows_x64.sh#L15

cavearr, 2024-10-27:
Hi @zapta this utility is needed for some manufacturers. You can’t remove it

zapta, 2024-10-27:
Sounds good.

zapta, 2024-11-05:
@cavearr, what do you think about moving the ftdi_eeprom utility out of the oss-cad-suite package and adding the tools-system packages as yet another apio supported package?  This will provide the ftdi_eeprom utility (and others) for production and other purposes without having to pigi-back on the oss-cad-suite package.

(The oss-cad-suite builder currently takes the ftdi_eeprom utility from the tools-system package).

https://github.com/FPGAwars/tools-system/releases/tag/v1.1.2

cavearr, 2024-11-07:
Hello @zapta, first sorry for my delay. I live in Valencia (Spain) and the Dana situation has slowed everything down.

I think the same as you, but I proposed something similar a long time ago and I didn't like it very much or it didn't fit at the time :( . If you elaborate the proposal further (package organization, new repository if necessary, etc) it might be time :) , You have my 👍 on this.

zapta, 2024-11-07:
Hi @cavearr, hope everything will go well with the storm. My thinking is to use the existing tools-systems repo and package  https://github.com/FPGAwars/tools-system , adding it to apio's list of packages (at least for the platforms that currently has ftdi_eeprom), and modify the build script of tools-oss-cad-suite to stop copying ftdi_eeprom to oss-cad-suite.  I will test it before submitting any change.

Does this make sense?

https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/scripts/install_windows_x64.sh#L15

cavearr, 2024-11-07:
Yes, i'm fine ! thanks a lot!  but the situation is very hard.

For me go ahead! and if someone complains, we'll see what we do.

zapta, 2024-11-12:
Question answered. Closing.  Will change the package logic to load the system-tools separately from oss-cad-suite. This should not impact prod.

closed 2024-11-12 by zapta