FPGAwars/apio #448: tools-oss-tools-cad nextptr_ice40 template interfers with upstream functionality

zapta, 2024-10-27:
@Obijuan, @cavearr, a question for you. The build script of the tools-oss-cad-suite package replace the original bin/nextptr_ice40 file with a simplified version from the template directory, which seems to interfer with the functionality of nextpnr. For example, the --gui flag does work with the original script but not with the replacement.

Any thought of moving back to the original scripts?  This will also simplify propogating nextpnr issues to the nextpnr team.

https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/scripts/install_darwin_arm64.sh#L66

Original nextptr_ice40
```
#!/usr/bin/env bash
release_bindir="$(dirname "${BASH_SOURCE[0]}")"
release_bindir_abs="$("$release_bindir"/../libexec/realpath "$release_bindir")"
release_topdir_abs="$("$release_bindir"/../libexec/realpath "$release_bindir/..")"
export PATH="$release_bindir_abs:$PATH"
export PYTHONHOME="$release_topdir_abs"
export PYTHONNOUSERSITE=1
export SSL_CERT_FILE="$release_topdir_abs"/etc/cacert.pem
export QT_PLUGIN_PATH="$release_topdir_abs/lib/qt5/plugins"
export QT_LOGGING_RULES="*=false"
unset QT_QPA_PLATFORMTHEME
unset QT_STYLE_OVERRIDE
export XDG_DATA_DIRS="$release_topdir_abs"/share
export XDG_CONFIG_DIRS="$release_topdir_abs"
export XDG_CONFIG_HOME=$HOME/.config/yosyshq
export XDG_CACHE_HOME=$HOME/.cache/yosyshq
export XDG_DATA_HOME=$HOME/.local/share/yosyshq
export LC_ALL="C"
mkdir -p $HOME/.config/yosyshq $HOME/.local/share/yosyshq
exec "$release_topdir_abs"/libexec/nextpnr-ice40 "$@"
```

Replacement nextpnr_ice40
```
#!/usr/bin/env bash
release_bindir="$(dirname "${BASH_SOURCE[0]}")"
release_bindir_abs="$("$release_bindir"/../libexec/realpath "$release_bindir")"
release_topdir_abs="$("$release_bindir"/../libexec/realpath "$release_bindir/..")"
export PATH="$release_bindir_abs:$PATH"
exec "$release_topdir_abs"/libexec/nextpnr-ice40 "$@"
```

cavearr, 2024-10-27:
For me, go ahead with the  update with the original script, but i don't know if this crashes with one of the @Obijuan works.

zapta, 2024-11-12:
Switched to the new oss-cad-suite package model. Closing.

closed 2024-11-12 by zapta