FPGAwars/apio #449: Please create a FPGAwars/tools-graphviz repository.

zapta, 2024-11-02:
@Obijuan, @cavearr, please create a ``FPGAWars/tools-graphviz`` repository for the graphviz package. You can copy the files and release from here https://github.com/zapta/tools-graphviz (but please just copy, do no fork, since my repo is temporary and will go away). 

I am working on having the ``gtkwave`` and ``dot`` commands being installed by apio packages on all three platforms (for ``apio sim``, and ``apio graph`` respectively).  The upstream oss-cad-tools packages already includes the graphviz tools for linux and darwin but not for windows. This package fills this gap. I already tested it on a windows machine.

cavearr, 2024-11-02:
Here it is! https://github.com/FPGAwars/tools-graphviz

 I have created the skeleton for you, I prefer that you upload it, you are the architect of this package and it is a great job, you should be the main contributor.

Thank you so much!

cavearr, 2024-11-02:
@zapta 🔥

zapta, 2024-11-02:
Thanks Carlos!  Will do.

zapta, 2024-11-12:
Release created. Closing.

closed 2024-11-12 by zapta