FPGAwars/apio #450: iceprog.exe doesn't upload in later versions of oss-cad-suite

zapta, 2024-11-09:
NOTE: I was able to reproduce this problem with upstream oss-cad-suite packages from YosysHQ , such that this is not apio specific.

On my windows box 'apio upload' fails with the apio dev version but not with apio prod version. The failure is in the iceprog.exe stage which exits with the following error.

```
iceprog -d i:0x0403:0x6010:0 hardware.bin
scons: *** [upload] Error 3221225477
```

The problem seems to be in the oss-cad-suite packages and was reproduced independently also without apio.  This one which is also used for the apio prod version works https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-10-03/oss-cad-suite-windows-x64-20231003.exe but newer versions don't.  The upstream package seems to have different versions of iceprog.exe in different versions. Below is a python script I used to hash the files.

I am using AMD/Ryzen, Windows 10, Alhambra II.

```python
# Print size and checksum of file(s).
#
# Usage:
#  python hash.py .apio/packages/tools-oss-cad-suite/bin/*.exe

import hashlib
import argparse
import os.path
from pathlib import Path

parser = argparse.ArgumentParser()
parser.add_argument('files', nargs='+')
args = parser.parse_args()

for fname in args.files:
  basename = os.path.basename(fname)
  if not os.path.isfile(fname):
      print(f"{basename:40} (not a file)")
      continue

  file_contents = Path(fname).read_bytes()
  md5_hash = hashlib.md5(file_contents).hexdigest()
  size = os.path.getsize(fname)
  print(f"{basename:40} {size:12}  {md5_hash}")
```

cavearr, 2024-11-12:
@zapta , i'm not a windows user, but with the newer oss cad suite , the ftdi lib has been updated and you need to use the native windows usb library (not replace with zadig and if you replace before, restore with zadig the original windows driver).

zapta, 2024-11-13:
@cavearr, I tried to go back to the original ftdi drivers and program directly with ``iceprog`` but it still doesn't work, details below. @Obijuan, any idea how to program Alhambra II on windows with the newer ``oss-cad-suite`` packages?

Channel 0 has the FTDI driver:
[image]

Channel 1 has the FTDI driver:
[image]

Iceprog commands fail:
```
~/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED$ apio build
Setting the envinronment.
[Wed Nov 13 07:48:35 2024] Processing alhambra-ii
-------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top main -json hardware.json" -q main.v
Warning: wire '\q' is assigned in a block at main.v:3171.14-3171.21.
Warning: wire '\q' is assigned in a block at main.v:3175.6-3175.12.
nextpnr-ice40 --hx8k --package tq144:4k --json hardware.json --asc hardware.asc --report hardware.pnr --pcf main.pcf -q
icepack hardware.asc hardware.bin
==================================================== [SUCCESS] Took 6.61 seconds ====================================================
~/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED$ apio raw "iceprog -d i:0x0403:0x6010:0 hardware.bin"
Setting the envinronment.
init..
Can't find iCE FTDI USB device (device string i:0x0403:0x6010:0).
ABORT.
~/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED$ apio raw "iceprog  hardware.bin"
Setting the envinronment.
init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
ABORT.
~/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED$ apio system --lsusb
Setting the envinronment.
2109:0817 (bus 2, device 4) path: 8.3
1022:15e1 (bus 1, device 0)
046d:c52f (bus 2, device 7) path: 4.3.3
1022:15e0 (bus 2, device 0)
1038:1618 (bus 2, device 8) path: 4.3.2
0b0e:0305 (bus 2, device 6) path: 4.4
2109:2817 (bus 2, device 3) path: 4
2109:0817 (bus 2, device 1) path: 8
8087:0a2a (bus 1, device 1) path: 2
0403:6010 (bus 2, device 2) path: 2
2109:2817 (bus 2, device 5) path: 4.3
~/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED$
```

cavearr, 2024-11-13:
Hi @zapta yesterday i'll try with a friend and this works in his windows. I'll try to reproduce and send you and screenshot.

zapta, 2024-11-13:
@cavearr, It's working now. The driver to use is WinUSB.  This driver still needs zadig settings, right? If I uninstall the driver, next time Windows assigns the FTDIBUS driver, so I presume WinUSB is not the default driver.

[image]

cavearr, 2024-11-13:
Yes that is! i'm forget tell you that you need to plug and unplug usb after reinstall the driver, sorry!

For people that is the first time Icestudio installation, don't need to install any driver with zadig, i'm removing this step from icestudio, all works with windows native driver.

zapta, 2024-11-13:
Carlos,

What driver do you see in zadig when you install alhambra II on a fresh windows system?

Do we still need  ``apio drivers --enable-ftdi`` and ``apio drivers --disable-ftd`` on windows?  If so, how their instruction text should be changed, just replace 'libusb' with 'winusb' ?

https://github.com/FPGAwars/apio/blob/f74eea6163964353e63addca804de4026d64e4f0/apio/managers/drivers.py#L18

https://github.com/FPGAwars/apio/blob/f74eea6163964353e63addca804de4026d64e4f0/apio/managers/drivers.py#L48

cavearr, 2024-11-13:
In a fresh windows os you don't need to install any driver. works natively.  I don't try on winows this two commands, if you give me until tomorrow afternoon , in the morning i could test in a fresh windows computer and tell you the result of this tests.

Tell me what commands do you need to try and in ther morning i'll do it.

zapta, 2024-11-13:
Hi @cavearr, please try this and post here how it went.

Prerequisites:  
* A fresh windows machine.
* The latest apio dev version.
*  An Alhambra II board connected to the comptuer.

1. Open Alhambra II in zadig and let me know what drivers are listed for interface 0 and for interface 1.

2. Decide if the command ``apio drivers --ftdi-enable`` is still needed for windows.  If the answer is no, do nothing and I will remove it for windows. If the answer is yes, run it and tell me how the instruction text should be changed.

3. Try to upload a project to Alhambra and see if it works.  ``apio upload``.

4. Decide if the command ``apio drivers --ftdi-disable`` is still needed for windows.  If the answer is no, do nothing and I will remove it for windows. If the answer is yes, run it and tell me how the instruction text should be changed.

cavearr, 2024-11-16:
Hi! confirmed by me and @Democrito in different computers with Windows 10 and 11 , is needed to install in Microsoft Windows the WinUSB driver in Alhambra board  port 0

If previously the user has been installed libusk , this should be replaced by the WinUSB driver.

zapta, 2024-11-17:
Thanks @cavearr, this is very useful!

I will make the change.

zapta, 2024-11-26:
Done. Now instructing the user to install WinUsb instead of libusbk.

https://github.com/FPGAwars/apio/blob/2d6b5994db700ad2dbab18e6af9194f6dce5413b/apio/managers/drivers.py#L33

Closing.

closed 2024-11-26 by zapta