FPGAwars/apio #452: Allow to control the order of the source files.

zapta, 2024-11-12:
Moved from https://github.com/FPGAwars/apio/issues/176#issuecomment-968181202

"I also find this request useful. It would be also nice if there is a way to specify the source files order. The current order seems to be alphabetical, which may not work in some cases. An example is building picosoc. One needs to list picosoc.v before picorv32.v so one cannot use apio build as is. The current workarounds are using Makefile or using apio raw or modifying SConstruct script or renaming the original files. Maybe having a possibility to have an ordered list of all sources in a config file could be beneficial. If the ordered list of sources is missing then existing alphabetical order of all sources could work."

fmodesto, 2024-12-07:
Could it be possible to allow more complex project structure? not only file ordering. There is an open request #227 to allow using several subdirectories.

In my project I have `src` and `test` directories. It would be great if apio could support specifying where to find the files.

For example in `apio.ini` 
```
synth_srcs = src/ lib/
test_srcs = test/
```

Or if required to control the order of files
```
synth_srcs = picosoc.v picorv32.v
```

Then when apio is searching for the source and test files. It can check that properties. If they are not specify it should default to the current directory as it's doing now. And if specify it could check if there are files or directories and build from there.

I have modify the bellow function to, for now, hardcode some values. I can run `apio build`, `apio test`, `apio sim test/leds.v` and works fine. It even creates the directories inside the `_build` dir. Only running a specific test is failing `apio test test/leds_tb.v`.

```
def get_source_files(env: SConsEnvironment) -> Tuple[List[str], List[str]]:
    source_files: List[File] = env.Glob("src/*.v")
    test_files: List[File] = env.Glob("test/*.v")

    synth_srcs = []
    test_srcs = []

    # test might contain non test bench like global utilities, mock components...
    # so iterate each group individually
    basedir: File = env.Dir('#') # FIXME is there a better way?
    for file in source_files:
        if not has_testbench_name(env, file.name):
            synth_srcs.append(basedir.rel_path(file))
    for file in test_files:
        if has_testbench_name(env, file.name):
            test_srcs.append(basedir.rel_path(file))

    return (synth_srcs, test_srcs)
```

zapta, 2024-12-07:
Thanks @fmodesto, this is a very good idea. Generally speaking we follow platformio conventions and it does support separate src/lib/test directories and nest directories.

In the next release we plan to have these initial steps:
1. Artifact files such as hardware.json are now placed in a separated subdirectory (``_build`` vs platformio's ``.build``), rather than the source directory.
2. The file apio.ini will be mandatory.  (currently it's not hence the flags such as --board in 'apio build').

Once apio.ini will be mandatory, we will be in a better place to add project options.

zapta, 2025-05-07:
The new PR https://github.com/FPGAwars/apio/pull/591 provides a workaround that allows the user to control the files order. Source files are now sorted by their path relative to the project root, and as a result, user can control the order by controlling directories and files. name. 

Consider the project structure below:
```
project
    apio.ini
    zzz.v
    src0
        bbb.v
        ccc.v
    src1
        aaa.v
        ddd.v
```

The files will be listed in the apio commands as

```
zzz.v src0/bbb.v src0/ccc.v src1/aaa.v src1/ddd.v
```

That's is a lexicographic order of the parent directories names and then in each directory a lexicographic order of the file names. The order is case insensitive.

Considering that this workaround is available and that there was no high demand to this feature, we don't plan at this point to provide a more explicit support (e.g. using new options in apio.ini).

closed 2025-05-07 by zapta