FPGAwars/apio #460: A question regarding the package installation platform id

zapta, 2024-11-14:
@Obijuan , @cavearr ,  I am trying to understand how platform ids are mapped to apio packages download URL.  

Looking at the code below, it creates a list with two entries, each with a download URLs,  entry [0] has the full platform id, e.g. ``darwin_arm64`` and entry [1] has the base platform id, e.g, ``darwin``.

My question is, is the second entry used anywhere, e.g. as a fallback if the first URL doesn't exist?  I can't couldn't find where it's used.

https://github.com/FPGAwars/apio/blob/f74eea6163964353e63addca804de4026d64e4f0/apio/managers/installer.py#L141

zapta, 2024-11-14:
@Obijuan, @cavearr, I am trying to understand the discrepancy between the 10 platform ids defined in the apio code and the 5 platform ids available in the oss-cad-suite release.  I speculate that there is somewhere a many-to-one  mapping between the two but couldn't find it, which leads me to think that we support only 5 platform ids.  Is it so?

```
# -- AVAILABLE PLATFORMS
PLATFORMS = [
    "linux",
    "linux_x86_64",
    "linux_i686",
    "linux_armv7l",
    "linux_aarch64",
    "windows",
    "windows_x86",
    "windows_amd64",
    "darwin",
    "darwin_arm64",
]
```

[image]

cavearr, 2024-11-14:
Hi @zapta! i'm not deep knowledge in apio, i only help with little things but i think this is duplicated or maybe could be 32bit platforms.

But if now oss-cad-suite don't support 32 bit platforms, i think could be better remove from apio (this is non sense).

i think the mapping is:

linux_i686. => intel 32 bits 
linux_x86_64 => intel 64bits
linux _armv7 => arm 32bits (maybe rpi 3B or olders than 4? 
windows_x86 => intel 32bits
windows__amd64 => intel 64bits
darwin => osx intel chips
darwin_arm64 => osx arm chips (current generation).

"linux" and "windows" definitions , i don't know what architecture define, or maybe could be a "generic" definition.

I think we could remove 32bits archs and linux and windows generics or almost pointed to most current used (intel 64 bits)

cavearr, 2024-11-14:
i'm forget linux_aarch64 for arm 64 bits that we need to maintain

zapta, 2024-11-14:
@cavearr, is the table below correct?  That is, only 5 platform ids, and 1:1 mapping between reported platform ids and the platform id in package release file names?

  

[image]

cavearr, 2024-11-14:
@zapta thanks for the table! give me the afternoon to check "linux" and "windows" both are my doubts what are these.

In other way, i'm going for the windows machines now (this morning was impossible), in the afternoon/night report you with the tests.

zapta, 2024-11-15:
Hi @cavearr, I plan to add to apio a new resource file platforms.json that maps platform ids to their attributes, for example, what package release file to select. Latter we can add more attributes. 

Please review and  let me if it's ok.

```
{
  "darwin": {
    "description": "Mac OSX, x86 64 bit",
    "package_id": "darwin"
  },
  "darwin_arm64": {
    "description": "Mac OSX, ARM 64 bit",
    "package_id": "darwin_arm64"
  },
  "linux_aarch64": {
    "description": "Linux ARM 64 bit",
    "package_id": "linux_aarch64"
  },
  "linux_x86_64": {
    "description": "Linux X86 64 bit",
    "package_id": "linux_x86_64"
  },
  "windows_amd64": {
    "description": "Windows x86 64 bit",
    "package_id": "windows_amd64"
  }
}
```

zapta, 2024-11-26:
Done. The only supported platform ids are now specified here

https://github.com/FPGAwars/apio/blob/develop/apio/resources/platforms.json

Closing.

closed 2024-11-26 by zapta