FPGAwars/apio #461: Is the board platform restriction still applicable?

zapta, 2024-11-15:
@Obijuan, @cavearr, there are two boards in boards.json that have platform restriction. The platform linux_armv7l doesn't seem to be supported anymore (see https://github.com/FPGAwars/apio/issues/460#issuecomment-2476865815).  Can these two boards be removed from boards.json()?

https://github.com/FPGAwars/apio/blob/d2c5bb3daaa40037b6e9b157aebec2862dfebd06/apio/resources/boards.json#L105

```json
  "icoboard": {
    "name": "icoBOARD 1.0",
    "fpga": "iCE40-HX8K-CT256",
    "programmer": {
      "type": "icoprog"
    },
    "platform": "linux_armv7l"     <<-----
  },
  "Cat-board": {
    "name": "CAT Board",
    "fpga": "iCE40-HX8K-CT256",
    "programmer": {
      "type": "litterbox"
    },
    "platform": "linux_armv7l"     <<-----
  },
```

cavearr, 2024-11-15:
@zapta, I think if icoprog or litterbox existed in linux_aarch64 we should update it.

I will look for these tools, I don't know these boards, but I will try to locate them and see if we can compile them or they exist in the arm64 version for Linux.

In any case, since oss-cad-suite does not support this architecture, if programmers did not exist we would have to eliminate them, because they will really be obsolete boards.

zapta, 2024-11-17:
Hi @Obijuan, sounds good, I understand the need to preserve them.

Since the new apio release, allows using custom ``boards.json`` and ``programmers.json`` files in the project dir, how about adding to ``apio-examples`` repo an example for those boards that demonstrates it?. And if the user will find one day a programmer binary that works, they can place it on the path and run ``apio upload`` seamlessly.

This approach has two  benefit:
* The users will get a better starting point and documentation of the challenges using these boards.
* The official boards.json will not contain boards that are known to be non functional.

Any thoughts?

zapta, 2024-11-20:
I moved this to a new issue here https://github.com/FPGAwars/apio/issues/470

Closing this one.

closed 2024-11-20 by zapta