FPGAwars/apio #483: [Feature request] Have uniform board names.

zapta, 2024-12-02:
@Obijuan , @cavearr, what do you think?

Currently there are a few issues related to board names

* Some board names use mixed case while other use all lower case. E.g. ``ECP5-Mini-12`` vs ``iceblink40-hx1k``.
* Some boards contains character that don't play well in directory names.  E.g. ``iCESugar-Pro_(FT2232H)``
* The directory names of some examples doesn't match the ``board=`` value in their ``apio.ini`` file. 

The proposal here is to
1. Define for each board a 'new name' using the code snippet below.
2. For some transition time support both the new names and old names, with a warning to the user if an old name if used.
3. Change the directory names of the apio examples to match their ``board=`` value in ``apio.ini``

Generating new names from old name:
```
 name = name.lower()
 name = name.replace("_", "-")
 name = name.replace("(", "")
 name = name.replace(")", "")
```

Old board names (left) vs new board names (right):
```
Alchitry-Cu                         -> alchitry-cu
alhambra-ii                         -> alhambra-ii
arice1                              -> arice1
blackice                            -> blackice
blackice-ii                         -> blackice-ii
blackice-mx                         -> blackice-mx
Butterstick-r10-2g-85k              -> butterstick-r10-2g-85k
Butterstick-r10-2g-85k_(FT2232H)    -> butterstick-r10-2g-85k-ft2232h
Butterstick-r10-2g-85k_(FT232H)     -> butterstick-r10-2g-85k-ft232h
Cat-board                           -> cat-board
ColorLight-5A-75B-V61               -> colorlight-5a-75b-v61
ColorLight-5A-75B-V7                -> colorlight-5a-75b-v7
ColorLight-5A-75B-V8                -> colorlight-5a-75b-v8
ColorLight-5A-75E-V6                -> colorlight-5a-75e-v6
ColorLight-5A-75E-V71_(FT2232H)     -> colorlight-5a-75e-v71-ft2232h
ColorLight-5A-75E-V71_(FT232H)      -> colorlight-5a-75e-v71-ft232h
ColorLight-5A-75E-V71_(USB-Blaster) -> colorlight-5a-75e-v71-usb-blaster
ColorLight-i5-v7.0_(FT2232H)        -> colorlight-i5-v7.0-ft2232h
ColorLight-i5-v7.0_(FT232H)         -> colorlight-i5-v7.0-ft232h
ColorLight-i5-v7.0_(USB-Blaster)    -> colorlight-i5-v7.0-usb-blaster
ColorLight-i9-v7.2_(FT2232H)        -> colorlight-i9-v7.2-ft2232h
ColorLight-i9-v7.2_(FT232H)         -> colorlight-i9-v7.2-ft232h
ColorLight-i9-v7.2_(USB-Blaster)    -> colorlight-i9-v7.2-usb-blaster
Cynthion-r1.4                       -> cynthion-r1.4
ECP5-Evaluation-Board               -> ecp5-evaluation-board
ECP5-Mini-12                        -> ecp5-mini-12
ECP5-Mini-25                        -> ecp5-mini-25
edu-ciaa-fpga                       -> edu-ciaa-fpga
FleaFPGA-Ohm_(FT2232H)              -> fleafpga-ohm-ft2232h
FleaFPGA-Ohm_(FT232H)               -> fleafpga-ohm-ft232h
FleaFPGA-Ohm_(USB-Blaster)          -> fleafpga-ohm-usb-blaster
fomu                                -> fomu
fpga101                             -> fpga101
go-board                            -> go-board
iCE40-HX1K-EVB                      -> ice40-hx1k-evb
iCE40-HX8K                          -> ice40-hx8k
iCE40-HX8K-EVB                      -> ice40-hx8k-evb
iCE40-UL1K-Breakout                 -> ice40-ul1k-breakout
iCE40-UP5K                          -> ice40-up5k
iceblink40-hx1k                     -> iceblink40-hx1k
iCEBreaker                          -> icebreaker
iCEBreaker-bitsy0                   -> icebreaker-bitsy0
iCEBreaker-bitsy1                   -> icebreaker-bitsy1
icefun                              -> icefun
icestick                            -> icestick
iCESugar-nano                       -> icesugar-nano
iCESugar-Pro_(FT2232H)              -> icesugar-pro-ft2232h
iCESugar-Pro_(FT232H)               -> icesugar-pro-ft232h
iCESugar-Pro_(USB-Blaster)          -> icesugar-pro-usb-blaster
iCESugar_1_5                        -> icesugar-1-5
iceWerx                             -> icewerx
icezum                              -> icezum
icoboard                            -> icoboard
kefir                               -> kefir
ODT_IcyBlue_Feather                 -> odt-icyblue-feather
ODT_RPGA_Feather                    -> odt-rpga-feather
OK-iCE40Pro                         -> ok-ice40pro
orangecrab-r02-25f                  -> orangecrab-r02-25f
orangecrab-r02-85f                  -> orangecrab-r02-85f
pi-sicle                            -> pi-sicle
pico-ice                            -> pico-ice
Sipeed-Tang-Nano                    -> sipeed-tang-nano
Sipeed-Tang-Nano-1k                 -> sipeed-tang-nano-1k
Sipeed-Tang-Nano-20k                -> sipeed-tang-nano-20k
Sipeed-Tang-Nano-4k                 -> sipeed-tang-nano-4k
Sipeed-Tang-Nano-9k                 -> sipeed-tang-nano-9k
ThetaMachines-ETH4K                 -> thetamachines-eth4k
TinyFPGA-B2                         -> tinyfpga-b2
TinyFPGA-BX                         -> tinyfpga-bx
TinyFPGA-EX-rev1                    -> tinyfpga-ex-rev1
TinyFPGA-EX-rev2                    -> tinyfpga-ex-rev2
ulx3s-12f                           -> ulx3s-12f
ulx3s-25f                           -> ulx3s-25f
ulx3s-45f                           -> ulx3s-45f
ulx3s-85f                           -> ulx3s-85f
upduino                             -> upduino
upduino2                            -> upduino2
upduino21                           -> upduino21
upduino3                            -> upduino3
upduino31                           -> upduino31
versa                               -> versa
```

cavearr, 2024-12-02:
Hi @zapta! i'm of course with you in the way that we need to standardize the board names (i'll do the same in Icestudio)

The only thing i think could be better is that we could define a mode of naming that improve the readability. We could use something like BEM in css , dividing the name in groups with -- (two lines for) kind of information and __ (two underline) for subtype or information less important, for example:

colorlight-5a-75e-v71-usb-blaster

convert into:
colorlight--5a__75e--v7--usb-blaster
boardname--model--version--other-info

Other example:

sipeed-tang-nano-20k
sipeed--tang-nano--20k

It's onlny a porposal but i think the readibility is better (i don't like long names all separeted with - but is my opinion).

zapta, 2024-12-02:
Hi @cavearr, we can use the convention of '--' to separate between fields but I am not sure where to insert them.  

Attached below is the names.txt file. You can download and edit it any way you like and post it back here.  The new name doesn't have to be an exact match of the old name, and it can be anything you want.

[names.txt](https://github.com/user-attachments/files/17982488/names.txt)

zapta, 2024-12-02:
Hi @cavearr, I think that the board id ``colorlight-i9-v72-ft2232h`` is clear enough without the concept of 'fields' but maybe you can improve other things in some board ids.

Please remember that this is the board id and that the boards.json has another field for a more human readable board description https://github.com/FPGAwars/apio/blob/e5d834bd102b91dc38e91bebb560e04fe3e0ec8d/apio/resources/boards.json#L9

cavearr, 2024-12-02:
@zapta Yes, yes don't worry, ok then continue with your proposal, don't complex the naming.

Thanks!

zapta, 2024-12-02:
@cavearr, sounds good. Once we have the mechanism for new/legacy names, we can change later.

zapta, 2024-12-05:
The new board naming scheme is implemented and PR sent.  Closing.

```
/projects/apio-dev/repo$ apio boards | cat
alchitry-cu                       |  (FPGA:iCE40-HX8K-CB132, ice40, hx, 8k, cb132)
alhambra-ii                       |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
arice1                            |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
blackice                          |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
blackice-ii                       |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
blackice-mx                       |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
butterstick-r10-2g-85k            |  (FPGA:ECP5-LFE5UM5G-85F-CABGA381, ecp5, um5g-85k, 85k, CABGA381)
butterstick-r10-2g-85k-ft2232h    |  (FPGA:ECP5-LFE5UM5G-85F-CABGA381, ecp5, um5g-85k, 85k, CABGA381)
butterstick-r10-2g-85k-ft232h     |  (FPGA:ECP5-LFE5UM5G-85F-CABGA381, ecp5, um5g-85k, 85k, CABGA381)
cat-board                         |  (FPGA:iCE40-HX8K-CT256, ice40, hx, 8k, ct256)
colorlight-5a-75b-v61             |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
colorlight-5a-75b-v7              |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-5a-75b-v8              |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-5a-75e-v6              |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-5a-75e-v71-ft2232h     |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-5a-75e-v71-ft232h      |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-5a-75e-v71-usb-blaster |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
colorlight-i5-v7-0-ft2232h        |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
colorlight-i5-v7-0-ft232h         |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
colorlight-i5-v7-0-usb-blaster    |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
colorlight-i9-v7-2-ft2232h        |  (FPGA:ECP5-LFE5U-45F-CABGA381, ecp5, 45k, 45k, CABGA381)
colorlight-i9-v7-2-ft232h         |  (FPGA:ECP5-LFE5U-45F-CABGA381, ecp5, 45k, 45k, CABGA381)
colorlight-i9-v7-2-usb-blaster    |  (FPGA:ECP5-LFE5U-45F-CABGA381, ecp5, 45k, 45k, CABGA381)
cynthion-r1-4                     |  (FPGA:ECP5-LFE5U-12F-CABGA256, ecp5, 12k, 12k, CABGA256)
ecp5-evaluation-board             |  (FPGA:ECP5-LFE5UM5G-85F-CABGA381, ecp5, um5g-85k, 85k, CABGA381)
ecp5-mini-12                      |  (FPGA:ECP5-LFE5U-12F-CABGA256, ecp5, 12k, 12k, CABGA256)
ecp5-mini-25                      |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
edu-ciaa-fpga                     |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
fleafpga-ohm-ft2232h              |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
fleafpga-ohm-ft232h               |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
fleafpga-ohm-usb-blaster          |  (FPGA:ECP5-LFE5U-25F-CABGA381, ecp5, 25k, 25k, CABGA381)
fomu                              |  (FPGA:iCE40-UP5K-UWG30, ice40, up, 5k, uwg30)
fpga101                           |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
go-board                          |  (FPGA:iCE40-HX1K-VQ100, ice40, hx, 1k, vq100)
ice40-hx1k-evb                    |  (FPGA:iCE40-HX1K-VQ100, ice40, hx, 1k, vq100)
ice40-hx8k                        |  (FPGA:iCE40-HX8K-CT256, ice40, hx, 8k, ct256)
ice40-hx8k-evb                    |  (FPGA:iCE40-HX8K-CT256, ice40, hx, 8k, ct256)
ice40-ul1k-breakout               |  (FPGA:iCE40-UL1K-CM36A, ice40, ul, 1k, cm36a)
ice40-up5k                        |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
iceblink40-hx1k                   |  (FPGA:iCE40-HX1K-VQ100, ice40, hx, 1k, vq100)
icebreaker                        |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
icebreaker-bitsy0                 |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
icebreaker-bitsy1                 |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
icefun                            |  (FPGA:iCE40-HX8K-CB132, ice40, hx, 8k, cb132)
icestick                          |  (FPGA:iCE40-HX1K-TQ144, ice40, hx, 1k, tq144)
icesugar-1-5                      |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
icesugar-nano                     |  (FPGA:iCE40-LP1K-CM36, ice40, lp, 1k, cm36)
icesugar-pro-ft2232h              |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
icesugar-pro-ft232h               |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
icesugar-pro-usb-blaster          |  (FPGA:ECP5-LFE5U-25F-CABGA256, ecp5, 25k, 25k, CABGA256)
icewerx                           |  (FPGA:iCE40-HX8K-CB132, ice40, hx, 8k, cb132)
icezum                            |  (FPGA:iCE40-HX1K-TQ144, ice40, hx, 1k, tq144)
icoboard                          |  (FPGA:iCE40-HX8K-CT256, ice40, hx, 8k, ct256)
kefir                             |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
odt-icyblue-feather               |  (FPGA:iCE40-U4K-SG48, ice40, u, 4k, sg48)
odt-rpga-feather                  |  (FPGA:iCE40-U4K-SG48, ice40, u, 4k, sg48)
ok-ice40pro                       |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
orangecrab-r02-25f                |  (FPGA:ECP5-LFE5U-25F-CSFBGA285, ecp5, 25k, 25k, CSFBGA285)
orangecrab-r02-85f                |  (FPGA:ECP5-LFE5U-85F-CSFBGA285, ecp5, 85k, 85k, CSFBGA285)
pi-sicle                          |  (FPGA:iCE40-HX4K-TQ144, ice40, hx, 8k, tq144:4k)
[... cut ...]
upduino                           |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
upduino2                          |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
upduino21                         |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
upduino3                          |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
upduino31                         |  (FPGA:iCE40-UP5K-SG48, ice40, up, 5k, sg48)
versa                             |  (FPGA:ECP5-LFE5UM-45F-CABGA381, ecp5, um-45k, 45k, CABGA381)
Total of 81 boards
```

closed 2024-12-05 by zapta