FPGAwars/apio #484: Unable to verify basic blinky on iCESugar-nano

YrnehEpoc, 2024-12-03:
I'm trying to follow[ Introduction to FPGA Part 2 - Getting Started with Yosys, IceStorm, and Apio | Digi-Key Electronics ](https://www.youtube.com/watch?v=gtkQ84Euyww&ab_channel=DigiKey) however I'm using the latest version of apio, 0.9.5 as well as Python 3.11.7 with an iCESugar-nano dev board. I'm trying to hold the yellow LED high on the board, which is on pin B6. my icesugar.pcf file looks like this: 
```
# For the iCESugar Board (iCE40UP5K-QFN48)

set_io D1 B6
```
my apio.ini file:
```
[env]
board = iCESugar-nano
top-module = blink_test
```
and my blink.v file:
```
module blink_test ( 
	output D1
);
 
 

assign D1 = 1'b1;

endmodule
```
when I call `apio verify` I get this in return
```
C:\Users\henry\Desktop\FPGA_Practice\test1\iCESugar_1_5\Blinky>apio verify
iverilog -o hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS "C:\Users\henry\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v" blink.v
scons: *** [hardware.out] Error 1
================================================ [ ERROR ] Took 0.21 seconds ================================================

C:\Users\henry\Desktop\FPGA_Practice\test1\iCESugar_1_5\Blinky>
```
I'm not sure how to interpret this error

zapta, 2024-12-03:
Can you build and verify the example iCESugar_1_5/Blinky?

Type this in an empty directory and it will copy the project files. Then
try apio build and apio verify and see if they work.

apio examples -f iCESugar_1_5/Blinky

On my system (mac) I get this (with apio Version: 0.9.5):

$ apio verify

iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o
hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS
"/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v"
blink.v

===============================================================================
[*SUCCESS*] Took 0.11 seconds

BTW, the command 'apio verify' is going to be deprecated and replaced by
'apio lint'.

zapta, 2024-12-03:
I was not able to reproduce on my win 10 machine. This below is a log of what I did and attached is a zip with the project directory.  Please try to unzip this project and see you can build or verify.

[test_project.zip](https://github.com/user-attachments/files/17998926/test_project.zip)

```
C:\Users\user\test_project>pip show apio
Name: apio
Version: 0.9.5
Summary: Open source ecosystem for open FPGA boards
Home-page: https://github.com/FPGAwars/apio
Author: Jesus Arroyo
Author-email: jesus.jkhlg@gmail.com
License:
Location: C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages
Requires: click, colorama, packaging, pyserial, requests, scons, semantic_version, wheel
Required-by:

C:\Users\user\test_project>python --version
Python 3.12.0

C:\Users\user\test_project>dir
 Volume in drive C has no label.
 Volume Serial Number is 0EAC-4058

 Directory of C:\Users\user\test_project

12/03/2024  12:50 PM    <DIR>          .
12/03/2024  12:50 PM    <DIR>          ..
12/03/2024  12:49 PM             4,030 .sconsign.dblite
12/03/2024  12:47 PM                52 apio.ini
12/03/2024  12:47 PM                69 blink_test.v
12/03/2024  12:47 PM                58 icesugar.pcf
               4 File(s)          4,209 bytes
               2 Dir(s)  95,072,772,096 bytes free

C:\Users\user\test_project>apio build
[Tue Dec  3 12:51:01 2024] Processing iCESugar-nano
-------------------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top blink_test -json hardware.json" -q blink_test.v
nextpnr-ice40 --lp1k --package cm36 --json hardware.json --asc hardware.asc --pcf icesugar.pcf -q
icepack hardware.asc hardware.bin
========================================================== [SUCCESS] Took 1.83 seconds ==========================================================

C:\Users\user\test_project>apio verify
iverilog -o hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS "C:\Users\user\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v" blink_test.v
========================================================== [SUCCESS] Took 0.68 seconds ==========================================================
```

YrnehEpoc, 2024-12-04:
I attempted to build and verify the blink apio example iCESugar_1_5/Blinky

this is the result:
```
C:\Users\henry\Downloads\test>apio examples -f iCESugar_1_5/Blinky
Copying iCESugar_1_5/Blinky example files ...
Example files 'iCESugar_1_5/Blinky' have been successfully created!

C:\Users\henry\Downloads\test>ls
apio.ini  blink.v  icesugar.pcf

C:\Users\henry\Downloads\test>apio lint
verilator --lint-only --timing -Wno-TIMESCALEMOD blink.v
scons: *** [hardware] Error 1
================================================ [ ERROR ] Took 0.21 seconds ================================================

C:\Users\henry\Downloads\test>apio verify
iverilog -o hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS "C:\Users\henry\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v" blink.v
scons: *** [hardware.out] Error 1
================================================ [ ERROR ] Took 0.20 seconds ================================================

C:\Users\henry\Downloads\test>
```

I also downloaded your example project zip, unzipped it and tried to build to no avail. The only obvious difference between our setups is that you're using python 3.12 and I'm using 3.11.7, though I don't think this is the source of the issue. Here's the results of building the example project
```
C:\Users\henry\Downloads\test_project>ls
apio.ini  blink_test.v  hardware.asc  hardware.bin  hardware.json  hardware.out  icesugar.pcf

C:\Users\henry\Downloads\test_project>apio build
[Tue Dec  3 21:34:30 2024] Processing iCESugar-nano
------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top blink_test -json hardware.json" -q blink_test.v
scons: *** [hardware.json] Error 1
================================================ [ ERROR ] Took 0.34 seconds ================================================
```

zapta, 2024-12-04:
I reinstalled apio from scratch and tried the project in the zip file I sent you.  ``apio build'' and '`apio verify`` succeeded but ``apio lint`` failed.  If ``apio lint`` and ``apio verify`` do not work for you, I suggest to continue without them, since ``apio build`` checks your code anyway.

BTW, that was fixed in the ongoing apio dev version (next apio release) and ``apio verify`` and ``apio lint`` were merged into a single command.

This is how I reinstalled apio from scratch
```

1. Use the windows explorer to delte the directory .apio in the users home directory, e.g. /Users/user/.apio.

2. Remove all pip packages

    pip freeze > temp.txt

    pip uninstall -r temp.txt -y
   
3. Verify that no pip packages are installed (list should be empty)

    pip freeze
	
4. Clear pip cache

    pip cache purge
	
5. Install the latest apio 

    pip install apio

6. Install apio packages

    apio install --all
```

My results with the project in the zip file
```
C:\Users\user\test_project>apio clean
Removed hardware.json
Removed hardware.asc
Removed hardware.bin
Removed hardware.out
========================================================== [SUCCESS] Took 0.45 seconds ==========================================================

C:\Users\user\test_project>apio build
[Wed Dec  4 11:38:12 2024] Processing iCESugar-nano
-------------------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top blink_test -json hardware.json" -q blink_test.v
nextpnr-ice40 --lp1k --package cm36 --json hardware.json --asc hardware.asc --pcf icesugar.pcf -q
icepack hardware.asc hardware.bin
========================================================== [SUCCESS] Took 5.14 seconds ==========================================================

C:\Users\user\test_project>apio lint
verilator --lint-only --timing -Wno-TIMESCALEMOD blink_test.v
'verilator' is not recognized as an internal or external command,
operable program or batch file.
scons: *** [hardware] Error 1
========================================================== [ ERROR ] Took 0.49 seconds ==========================================================

C:\Users\user\test_project>apio verify
iverilog -o hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS "C:\Users\user\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v" blink_test.v
========================================================== [SUCCESS] Took 1.31 seconds ==========================================================
```

zapta, 2024-12-22:
The issue was fixed and will be included in next release.

Closing.

closed 2024-12-22 by zapta