FPGAwars/apio #487: Can't follow quick start / no examples found

jpf91, 2024-12-06:
Hi,

I'm following the instructions at https://github.com/FPGAwars/apio/wiki/Quick-start#apio-installation but I can't get `apio example` to work. Here are the commands I'm using:

```bash
# Set up Ubuntu 22.04 container in distrobox
# Wiki says 20.04, but there the python version is too old
distrobox create --image ubuntu:22.04 --name apio
distrobox enter apio
```

```bash
# Install according to quick start
sudo apt install python3 python3-pip
python3 --version
Python 3.10.12

sudo pip3 install -U apio
apio install -a
```

```bash
# This will fail in the container, but that's expected. I hope it's not needed for simulating the examples?
apio drivers --ftdi-enable
cp: cannot create regular file '/etc/udev/rules.d/80-fpga-ftdi.rules': No such file or directory
sudo: udevadm: command not found
sudo: udevadm: command not found
udev: unrecognized service
FTDI drivers enabled
Unplug and reconnect your board
```

```bash
# apio doesn't find any example:
apio examples -d Alhambra-II/Blinky

Warning: this example does not exist
Use `apio examples -l` for listing all the available examples
```

```bash
# Let's have a look at the list:
apio examples -l

───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Total: 27

To get an example, use the command:
   apio examples -d/-f name

Example of use:
   apio examples -f leds
Copy the leds example files to the current directory
```

So for some reason apio thinks there are 27 eamples, but it can't list any. The example files are however installed and `tree` / `ls` also see them in the container:
```bash
tree ~/.apio/packages/examples/examples/
/var/home/jpfau/.apio/packages/examples/examples/
├── alchitry-cu
│   └── blinky
│       ├── apio.ini
│       ├── blinky.v
│       ├── info
│       └── pinout.pcf
├── alhambra-ii
│   ├── blinky
│   │   ├── apio.ini
│   │   ├── Blinky.v
│   │   ├── info
│   │   └── pinout.pcf
...
```

I'm not sure what could be causing this. `apio examples -d alhambra-ii/blinky` (lowercase) doesn't work either. Any idea how to debug this further?

zapta, 2024-12-06:
Hi @Obijuan, can you make these changes in apio-examples?

1. Create a new file named ``VERSION_DEV`` that contains ``0.1.1``
2. Change the content of the file VERSION to ``0.0.36``
3. Test with the prod apio that ``apio examples -l`` lists the examples.

zapta, 2024-12-06:
@Obijuan, the examples package seems to be broken, containing the entire repo structure. Can you rebuilt it by running the ``build.sh`` script It should create the proper .zip in the _package directory.  Also, I suggest having file ``VERSION`` pointing to 0.0.36 and ``VERSION_DEV`` to the latest release.

@jpf91, while we fix the packages, you can try moving the content of ``packages/examples/examples`` one level up to ``packages/examples`` and try again ``apio examples -l``.  Alternatively, you can copy download ``appio-examples-0.0.36.zip`` from here, upzip, and copy the files from the examples you like.  https://github.com/FPGAwars/apio-examples/releases/tag/0.0.36

jpf91, 2024-12-06:
Thank you for the quick reply :smile: I'll test next Monday when I'm back at work and will report my results here :+1:

zapta, 2024-12-06:
@Obijuan , I am updating the apio-example README. Here is a heads up regarding the release creation.

[image]

zapta, 2024-12-08:
Created two PRs that should fix this:

https://github.com/FPGAwars/apio-examples/pull/32

https://github.com/FPGAwars/apio/pull/491

zapta, 2024-12-18:
Fixed. Prod is working now and a PR was sent for dev.

closed 2024-12-18 by zapta

jpf91, 2024-12-18:
I can confirm it's working fine now. Thanks :+1:

zapta, 2024-12-18:
Thanks @jpf91.