FPGAwars/apio #492: [For discussion] Add an apio package for Veriable?

zapta, 2024-12-09:
@Obijuan, @cavearr, what do you think?  If you can create a repo ``FPGAWars/tools-verible``, I can populate it.

Verible is an open source project with multi platforms verilog tools, including, formatter, language server (e.g. for Visual Studio Code) and more. At first stage it can be installed as an apio package and used via the ``apio raw`` command, and later, once we gain confidence in it, via dedicated apio commands such as ``apio format`` to format verilog files.

I am using it for more than a year now on darwin_arm64 using the the VSC Verible extension, which requires the ``verible-verilog-format`` (formatter) and ``verible-verilog-ls`` (language server) to be available locally. 

Project home:
https://chipsalliance.github.io/verible/

Repository:
https://github.com/chipsalliance/verible

Package content:
[image]

Release assers:
[image]

cavearr, 2024-12-09:
Hi @zapta ! i use verible too! why do you want to fork this repo instead of using the original? I'll create without problem is more for understand the needings (i don't analyze the problem only ask you :) )

zapta, 2024-12-09:
Hi Carlos, for no specific reasons, just using the existing practices for
Apio packages, unpacking, repacking, and using a VERSION file. I don’t plan
to change the content of the package.

zapta, 2024-12-09:
We may also need to change the package name to comply with apio platform
ids and package name conventions.

cavearr, 2024-12-09:
Go ahead! the repositoy is created and you have full-access.

zapta, 2024-12-09:
Thanks Carlos!

zapta, 2024-12-10:
Hi @cavearr, I populated the tools-verible repo with a build script and created the first version.

On different topic, the prod example package is broken. Can you you accept these two PRs and follow the instructions regarding the new release?   

The problem is that the example package was created by zipping the entire repo instead of using the build.sh script, and this created a package with an extra 'examples' directory level.

https://github.com/FPGAwars/apio-examples/pull/32
https://github.com/FPGAwars/apio/pulls

zapta, 2024-12-18:
Done. Closing.

closed 2024-12-18 by zapta