FPGAwars/apio #493: [For discussion] Improvements Apio.ini

cavearr, 2024-12-12:
Add the ability in Apio.ini of add defines to use into verilog files.

For example:

[env]
board = Sipeed-Tang-Nano-1k
top-module = main
defines = 
    -KEY1=VALUE1
    -KEY2=VALUE2

This defines will be passed as yosys argument with -D option.

In other hand could be interesting the capability of passing defines in build command as argument for example:

apio build --defines KEY1=VALUE1,KEY2=VALUE2

In other way we could define multiple target boards to use the same file for multiple targets, for example:

[env]
board = Sipeed-Tang-Nano-1k
top-module = main

[env:Alhambra-ii]
defines = 
    -myBoard=ALHAMBRA

[env:Fomu]
defines = 
    -myBoard=FOMU
    -LED=RGB

We could do :

apio build    => This only will apply env block

or

apio buils --env Fomu.   => This will apply env + env:Fomu blocks

zapta, 2024-12-12:
@cavearr, @JohannesSchramm, these are very good ideas and in the spirit of platformio. The next step is making apio.ini required and the central source of project configuration, eliminating the configuration flags such as --board and --fpga. This will make adding new project controls much more maintainable.

@cavearr, this is related to item 2 here https://github.com/FPGAwars/apio/issues/453.  Any news related to items 2-5?

cavearr, 2024-12-12:
Hi @zapta today or tomorrow i'm finishing icestudio integration and testing and we could remove the flags.

About the config file i could do it if you have other tasks , count with me, if we decide go in this way i'm advance in Icestudio part to have an apio.ini editor to customize the apio.ini for icestudio projects.

Let's go team!

zapta, 2024-12-12:
Hi @cavearr, I can do it and already have a pending change for part of the works but am waiting for OK from you for items 2-5.

zapta, 2024-12-23:
@cavearr 
1. Please ignored my previous post. I was referring to a different feature.
2. Please rename this issue to **[Feature Request] Support multiple build env variations.**

cavearr, 2024-12-23:
Done!

zapta, 2025-05-08:
For reference, this is the documentation of the platformio .ini file. Some interesting features:

* Ability to specify explicitly the default env
* Ability to share options using implicit and explicit inheritance
* Blocks that contains option but are not envs.
* Values placeholders ($interoplation).

Also, the build output directory is env dependent such that if building 3 envs, there are three build directory at the same time. That is
```
project_dir
    _build
        env1
            <build artifacts>
        env2
            <build artifacts>
        env3 
            <build artifacts>

```

@cavearr, what do you think about the having a build directory that depends on the env name?  It adds unnecessary complexity to a single env user but is also more flexible.

cavearr, 2025-05-08:
@zapta, i think this is very interesting option, if you thinks the same for me is an interesting feature.

zapta, 2025-05-11:
@cavearr, the multi env is implemented and the -D key=val idea has now its own issue https://github.com/FPGAwars/apio/issues/599

Closing this one.