FPGAwars/apio #494: Install of the develop version fails under Windows with "Failed building RPi.GPIO"

JohannesSchramm, 2024-12-12:
Hi everyone,

i followed the [Guide for Apio development](https://github.com/FPGAwars/apio/wiki/Apio-development-version) and ran into an issue.

when i execute the command `flit install` the install fails with the following message:

```
.....
Building wheels for collected packages: apio, RPi.GPIO
  Building wheel for apio (pyproject.toml) ... done
  Created wheel for apio: filename=apio-0.9.6-py3-none-any.whl size=139194 sha256=de35c68d0dec3b48198fb3e43975033c2c61132a2c202331b8246973caaaae09
  Stored in directory: C:\Users\Johannes\AppData\Local\Temp\pip-ephem-wheel-cache-jecoknts\wheels\01\e6\33\5e481dcbf8a8f8d050d01223ec3a9354e3f0cc707657b078f2
  Building wheel for RPi.GPIO (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [18 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-3.9
      creating build\lib.win-amd64-3.9\RPi
      copying RPi\__init__.py -> build\lib.win-amd64-3.9\RPi
      creating build\lib.win-amd64-3.9\RPi\GPIO
      copying RPi\GPIO\__init__.py -> build\lib.win-amd64-3.9\RPi\GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build\temp.win-amd64-3.9
      creating build\temp.win-amd64-3.9\Release
      creating build\temp.win-amd64-3.9\Release\source
      C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.42.34433\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ie:\git\apio\venv\include -IC:\Users\Johannes\AppData\Local\Programs\Python\Python39\include -IC:\Users\Johannes\AppData\Local\Programs\Python\Python39\include -IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.42.34433\include -IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um -IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared -IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt -IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt /Tcsource/c_gpio.c /Fobuild\temp.win-amd64-3.9\Release\source/c_gpio.obj       
      c_gpio.c
      source/c_gpio.c(27): fatal error C1083: Datei (Include) kann nicht ge”ffnet werden: "sys/mman.h": No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.42.34433\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.       
  ERROR: Failed building wheel for RPi.GPIO
  Running setup.py clean for RPi.GPIO
Successfully built apio
Failed to build RPi.GPIO
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (RPi.GPIO)
Traceback (most recent call last):
  File "C:\Users\Johannes\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in 
_run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Johannes\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\Git\apio\venv\Scripts\flit.exe\__main__.py", line 7, in <module>
  File "e:\git\apio\venv\lib\site-packages\flit\__init__.py", line 228, in main
    installer.install()
  File "e:\git\apio\venv\lib\site-packages\flit\install.py", line 433, in install
    self.install_with_pip()
  File "e:\git\apio\venv\lib\site-packages\flit\install.py", line 371, in install_with_pip   
    check_call(cmd, shell=shell)
  File "C:\Users\Johannes\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['e:\\git\\apio\\venv\\scripts\\python.exe', '-m', 'pip', 'install', '.[tinyprog,icefunprog,litterbox,blackiceprog,tinyfpgab]']' returned non-zero exit status 1.
```

As I understand the build of the RPi.GPIO package fails, because it supports only linux and im using windows. 
If i remove the dependency of litterbox, where the dependency on RPi.GPIO comes from(line 38 of the pyproject.toml file), `flit install` runs sucessfully, but i have no idea what side effects this has.

It seems to me that the package litterbox is used by the litterbox programmer. So shouldn't be the python package litterbox be a dependency of the programmer and not of apio itself?

zapta, 2024-12-12:
@JohannesSchramm, I install the dev version of apio by brining a copy of the apio repo to my local drive (downloading .zip or using git clone) and then install it as the apio package using ``pip install -e .`` in the root directory of the repo. That worked for on mac osx and windows.  

There are a few other developers hints in the repo itself https://github.com/FPGAwars/apio/blob/develop/DEVELOPERS.md

JohannesSchramm, 2024-12-13:
@zapta, thanks for your hint. `pip install -e .` work for me. Does it the same as the `flit install` command, from the [Guide for Apio development](https://github.com/FPGAwars/apio/wiki/Apio-development-version)? 

And should the `flit install` not be used? If so it would be nice to update the wiki page.

zapta, 2024-12-13:
@JohannesSchramm, I am not familiar with flit and with that guide but worked for some time with ``pip install -e .`` and it works very well for me, both for apio developement and using the dev apio for my verilog project. I will that guide and see if we need to update it.

BTW, When the apio dev is installed via ``pip install -e .``, it's ok to uninstall it via ``pip uninstall apio``, it will not delete or touch your apio dev repository.

JohannesSchramm, 2024-12-13:
@zapta, then I will use the commands from the [DEVELOPERS.md](https://github.com/FPGAwars/apio/blob/develop/DEVELOPERS.md)  to test my local changes. 

The wiki page is probably outdated.

Thanks for your help.

closed 2024-12-13 by JohannesSchramm

zapta, 2024-12-14:
I updated the wiki page. Still needs some cleanup but the '-e .' is there.