FPGAwars/apio #495: [For discussion] ideal project structure

JohannesSchramm, 2024-12-12:
Hey everyone,

my current project structure looks like to following:

```
src
├── top
│   ├── top.v
│   └── top_tb.v
├── spi_register
│   ├── spi_register.v
│   └── spi_register_tb.v
├── clk_gen
│   ├── glk_gen.v
│   └── glk_gen_tb.v
├── apio.ini
├── constrains.pcf
├── config.yaml <- custom config file
└── SConstruct <- custom sconstruct file
```

So currently i use a custom sconstruct file with a custom config file to select the testbench used and to search for sourcefiles in the sub-directories. But the SConstruct file is quite old and doesn't support new features of apio, so i would like to migrate to an new SConstruct file (idealy to the default one).

So my question is what is the prefered way to structure an apio project? 
For me it would make sense to allow source files in sub-directories, what do you think of that?
Also should include macros be used to link the source files together?

I am interested in your feedback and would like to contribute to the apio project.

zapta, 2024-12-12:
Hi @JohannesSchramm

Supporting project specific SConstruct files is a feature that will be removed from the next release of APIO because of our inability to guarantee backward compatibility as apio evolves and the official SConstruct chagnes. Our new philosophy is to add the missing functionality to the standard apio commands.

As for supporting sub directories, this feature is on our short list and we are trying to figure out how it should look. Our current (tentative) thinking is not to dictate a specific directory structure as platformio does (with 'test', 'src' and 'lib' directories) but to allow any directory structure and just pick the verilog files and the testbench files cross the directories. 

If we will go with that appraoch, each user and team will be able to apply their preferences. E.g. for seperation between RTL and simulation files, or between core project files and library files. If you are looking for ideas I suggest to search the internet (e.g. here https://www.ece.ucdavis.edu/~bbaas/180/tutorials/file.organization.html ) or ask in relevant user groups (e.g. here https://www.eevblog.com/forum/fpga/a-question-for-hdl-developers-where-do-you-place-your-source-and-testbenches/ )

Also, if you want to hack SConstruct I suggest to make it in the apio dev version as it much more similar to the next apio release. To do it, you need to get a copy of the apio repository on your local drive (via zip download or git clone) and then do

```
rm ~/.apio     # delete the .apio directory under your user home directory. (use the command for your system)
pip uninstall apio   #  delete the old apio
cd <apio-dev-top-directory>   # Go to the apio top directory with pyproject.toml
pip install -e .    # Install apio from the current directory
apio packages --install     # Install all the packages for the new apio
```

JohannesSchramm, 2024-12-13:
Hey @zapta,

thanks for the links where project structures are discussed. Now it's also clear to me where the lack of support for sub directories comes from. I'm not sure which structure I prefer, so I appreciate the approach of letting the user decide how to structure the project. Also I like the idea to use a similar approach as plattformio does. 

E.g. the apio.ini options for the src files could look like this:
```
src_dir = src/
# to include sub-directories:
src_filter = 
  +<**/*.v>
  -<**/*_tb.v>
```

For simulation and testing the options could look similar:
```
test_dir = test/
# to include sub-directories:
test_filter = 
  +<**/*_tb.v>
```
And with the following option the user could set which testbench is used with `apio sim`:
```
sim_tb = test/my_tb.v
```

Another question, are there any arguments for or against the use of the include macro?

zapta, 2024-12-13:
@JohannesSchramm, if I got it correctly, you argue for apio supporting configurable directories types such as , source vs test.  If we will got with this approach we will need to define what is the semantic of each directory types (e.g. source are synthesizeable  and  test are unit tests that are simulated, one at a time, with all the synthesizeable files.  I am not a verilog expert so we will need to ask around.  BTW, I saw somewhere distinction between 'simulations' files and 'test' files, not sure what that mean.

As for macros inclusion, we have it in one of the apio examples, where the some test mechanics are abstracted in a macro file that can be shared by multiple tests. I think it's cleaner that way but I am not an expert. BTW, apio is aware of inclusion and will recompile the source file if the included file changes https://github.com/FPGAwars/apio/blob/f5c5df30ecccb0205112488994ecd4b952494b9f/apio/scons/scons_util.py#L365

https://github.com/FPGAwars/apio-examples/blob/master/examples/upduino31/testbench/main_tb.v#L4

zapta, 2024-12-13:
I created https://github.com/FPGAwars/apio/issues/497 for discussing the requirements.

JohannesSchramm, 2024-12-13:
@zapta, as we discussed it would be nice to allow the user to choose a directory structure. To allow different directories for synthesizeable files and testbenches, I thought of two options in the apio.ini file (src_dir and test_dir) to tell apio these directories. If someone wants to put all files in one directory, src_dir and test_dir could point to the same directory. 
Another idea would be to specify an option that accepts multiple directories, like the following:
```
dir =
 - src/
 - test/
```

I think apio currently uses the file extensions .v and _tb.v to distinguish between synthesizable and testbench files. This could be kept or a file filter like src_filter or test_filter (which I mentioned aboth) could be used.

Also, if include macros are used, I think only the top module needs to be specified to apio and all other files are found through the include (I tried this and it seems to work). Then the following would be enough:
```
top_file = top/top.v
```

These are just ideas of mine how this could be realised, so I think we can continue with this in https://github.com/FPGAwars/apio/issues/497.

My original question was how the project structure should be in apio, which you answered aboth.

closed 2024-12-13 by JohannesSchramm