FPGAwars/apio #499: [For discussion] Is APIO_PACKAGES_DIR (or APIO_PKG_DIR) needed?

zapta, 2024-12-17:
Currently apio dev support the env variables ``APIO_HOME_DIR`` and ``APIO_PACKAGES_DIR``.  Is ``APIO_PACKAGES_DIR`` used anywhere?  Can I simply have it hard wired to ``$APIO_HOME_DIR/packages``?  

Motivation is code and test simplicity.

@Obijuan, @cavearr, any thoughts?

cavearr, 2024-12-17:
Sorry @zapta but i don't have the deep knowledge in apio to decide if this envs could be reduced.

zapta, 2024-12-17:
@cavearr, you don't use in in icestudio, right?  I see only reference to APIO_HOME_DIR.

Obijuan, 2024-12-17:
> Currently apio dev support the env variables APIO_HOME_DIR and APIO_PACKAGES_DIR. Is APIO_PACKAGES_DIR used anywhere? Can I simply have it hard wired to $APIO_HOME_DIR/packages?
Yes, you can

cavearr, 2024-12-17:
no i don't use it 

Enviados: Martes, 17 de Diciembre 2024 23:08:33

zapta, 2024-12-18:
Thanks. Will delete it now. Closing.

closed 2024-12-18 by zapta