FPGAwars/apio #503: [Feature requested/Discussion] Support for iverilog at Apio lint

cavearr, 2024-12-18:
Hi! i'm migrating  Icestudio to verilator because now Icestudio "verify"not work ( until now , Icestudio spected iverilog output).

The change to iverilog appears me great but i think we should support boths in apio lint, defualt verilator but with a flag , use iverilog, what do you think team?

I want to know the reasons to move to verilator instead of iverilog @zapta, what are the points that moves you take this decision?

I think this is important this kind of decisions should be discussed.

Thanks for the work!

zapta, 2024-12-18:
Hi Carlos, if recall correctly verify didn’t detect some lint errors and it
was not obvious why do we need two lint commands and how to explain to
users the differences between them.

What is the problem with apio lint? Does it report errors that you would
like to ignore?  We can look into adding a —relax flag that will suppress
some error types.

If I recall correctly, icestudio generates lint issues. It would be nice to
fix them but if not possible we can suppress them.

(I am not near the computer, typing from my phone)

zapta, 2024-12-18:
I am not sure but does apio lint generate a file called
_build/hardware.vlt? If so, it contains examples of lint rules
suppressions.

cavearr, 2024-12-19:
@zapta, don't worry for ICestudio, i have been working with verilator now, i'm working testing step by step with a lot of boards and in some days all works perfecty, we are doing a good work! thanks for all @zapta !

This is only suggestion, because i think , thinking in a "big picture", there are users, that have implemented for example test rigs based on apio. Now, with the new syntax all rigs will be broken (angry users are comming or users that abandon apio for sure).

I think we need to plan a soft transiction, for example i think we need to maintain the deprecated commands almos one stable release and not remove it in this. Always with the message "This is a deprecated ......" and advise that in the next release this commands wil be removed.

In the same way, a lot of people could has scripts that parse the iverilog ouput generated by current apio and we shouldn't break them without prior advice.

About maintain verilator and verilog i tell you my thoughs.

I prefer verilator, i'm using a lot outside Icestudio and this is more adopted by industry. But in academic environmnets iverilog is used a lot. And as i tell you i know people that parse apio output inside actions, scripts, deploy tools,etc

the support for verilog-2005 is better in iverilog and although I like the analysis that verilator does better, iverilog verifies faster in large projects. for this i think It would be nice if, although verilator is the default mode for the linter, a flag like --linter=iverilog could be passed and when doing so, celery would use iverilog instead of verilator.

We do not have to explain anything about the use of one or the other, just give the possibility and the user with their knowledge and needs is the one who must decide.

In ICestudio it will initially use the default option of verilator, I don't know if I will give the configurator the option to select iverilog if we put it, initially I don't see any advantages, but as I told you I'm thinking with zoom and not just seeing celery as a monolithic tool and individually, but within a workflow in which it interacts with other processes.

This is my vision, but we must be in consensus, if you prefer to have a single option I would keep verilator.

Have a good day!

zapta, 2024-12-19:
Hi @cavearr, thanks for the feedback.

This release of apio changes a few basic things (e.g. making apio.ini required) so I hope that the simplicity and new functionality will compensate for the inconvenient  (we should make sure to describe it well in the release notes). 

Regarding lint, supporting two separate linter will make maintenance more complex and may be confusing to users. I think it's more efficient and clear to the users if we provide a single linter that works intuitively out of the box and supports some form of rule customization (e.g. using comment directives in the verilog code itsef, as we do with pylint, or by providing an optional linter config file in the project). 

Regarding the backward compatibility of the verify/lint functions, I would like to understand better what are the concerns. 

1. Is the  lint command is too strict compared to the verify command?  That is, too many errors or warnings.

2. Do people parse the output of `apio verify` and make sense of its error messages?  (I would expect that any automation needs to check just the exit code to determine success/failure)

3. The need to change the command line?

4. The new lint is significantly slower than verify?  If so, do we have an example of a large project that we can test with?

5. Something else?

BTW, users that will insist on linting using iverilog can use the workaround of using iverilog with the apio raw command since iverilog is still installed by default. Same for linting using verible.

Thanks,
zapta

zapta, 2024-12-19:
Hi @cavearr, I can change ``apio lint`` to use iverilog instead of verilator. Will this address your concern?

cavearr, 2024-12-19:
Noooo!! @zapta verilator is perfect and if i choose between both i prefer verilator!

but my question is if could be maintain iverilog with a flag that change verilator for verilog in this case.

If you view this hard to maintain, we advance with verilator way ;)

zapta, 2024-12-19:
Hi @cavearr, we can add to ``apio lint`` the ability to select a linter via a flag (apio installation already has by default three linter binaries, iverilog, verilator and verible).  The downside it that will be more complex to maintain (more permutations, different rules supressions mechanisms, etc) and will present the user with a dilemma which one to use.  

If you believe that there are good reasons to do it we can go that way. You have more experience with me with HDL design and understand the users.

BTW, my personal preference is verible because it provides a consistent set of linter, formatter, and a language server which we can use.

cavearr, 2024-12-19:
Hi @zapta! thanks a lot for your work is great!

i trust in your decisions 100%, only we need do more smooth the transition when we make changes.

For linter verible is great, if you want we could use it as default, i need take a decision because i'm working in the Icestudio parser for the linting. , VERIBLE?

ping me when you do it  ok?

Thans again!

zapta, 2024-12-19:
Hi @cavearr, I want to make sure I understand, do you suggest supporting now a single linter which will be verible?

I am OK with that.

zapta, 2024-12-19:
> Verible only for linter is my preferred option and i think we could
> maintain deprecated commands (verify with iverilog) for one release
> indicating that this is deprecated (like now).

Ok, I will do that. Thanks for the feedback.

zapta, 2024-12-19:
... icestudio will not use the deprecated ``apio verify`` right?

cavearr, 2024-12-19:
I'm migrating all to the new commands at develop, but for the momento i need to maintain the support of bots for the current apio stable release. 

In development i'm migrating to apio lint 

Enviados: Jueves, 19 de Diciembre 2024 19:08:26

cavearr, 2024-12-19:
Thanks @zapta, the deprecated commands only need to be alive for one more version, we can organize well the announcements and notices to users that the commands will be deprecated in the next one (we could release them in a couple of weeks or whenever you consider it and then wait for two or three months to release it again and then remove all the old commands, could be a possible plan, tell me what you think or what you have in mind, I would really appreciate it!

zapta, 2024-12-19:
We are talking just about the verify, right? Not the other way around four

zapta, 2024-12-19:
Carlos,  I can restore the verify command but not the other four. If you
want to restore them yourself I can point you to the relevant information
but I believe that this will take apio backwards. I was very transparent
with the plans for months so raising an objection only now is counter
productive. Please let me know how you want to proceed.

cavearr, 2024-12-19:
Everything is fine like this! Go ahead, thanks for continuing to review and for all the work, cheers up!

zapta, 2024-12-19:
Thanks Carlos!

cavearr, 2024-12-19:
tell me when have verilble working to do my final migration in Icestudio.

Now i have integrated verilator and check if you could, because i think in Windows we have som problem from Apio, i put yo an screenshot with apio in the terminal linting a simple example, although we use in short time verible, for simmulation you generate understand a similar command.

Try it in windows an view if the problem is replicated in your environment ( i'll  try windows 11 and @Democrito check in windoes 10 ) the same problem appears:
[image]
[01. One LED.zip](https://github.com/user-attachments/files/18203644/01.One.LED.zip)

zapta, 2024-12-19:
> @zapta <https://github.com/zapta> tell me when have verilble working to

Thanks Carlos, I will try to replicate this issue.

As for the verible linter, I am working now on the APIO_DEBUG env variable
and will work on verible after that.

zapta, 2024-12-20:
Hi Carlos, I sent this PR https://github.com/FPGAwars/apio/pull/504 with
the fix for 'apio lint' on windows.

zapta, 2024-12-22:
"I want to know the reasons to move to verilator instead of iverilog @zapta, what are the points that moves you take this decision?"

Carlos, we discuss it and the linter you prefer the most is verilator which is what implemented now.

If you have a change in mind such as switching to a different linter or adding a second linter, please file and issue with the details of you proposal such that it's actionable.

closed 2024-12-22 by zapta

cavearr, 2024-12-22:
Hi @zapta, all it's ok! this discussions is prior of the final decisions! we have duplicated, sorry , i think this issue could be close