FPGAwars/apio #505: [Feature Request] Switch 'apio lint' to use verible

zapta, 2024-12-20:
This came in a discussion with Carlos today, switching 'apio lint' from verilator to verible. Verible is already part of the standard packages that apio dev installs and it provides matching linter, formatter, and a language server. 

This issue focuses on the requirements and the implementation plans.

Requirements:
* 'apio lint' should work out of the box with no additional configuration or preparation.
* User will be able to customize the linter, e.g. enabling or disabling rules.
* Customization will be contained in the project such that sharing the project shares also the lint configuration.
* The lint will support both synthesizeable and testbenches, verilog (.v) and system-veilog files (.sv) files.
* Customization will be verible specific, that is, we are not going to hide or abstract the customization config (not reinventing the wheel).
* User will be able to lint only a portion of the project, e.g. by specifying one or more files or by specifying a module (?).

Implementation ideas
* 'apio lint' will issue command such as ``verible-verilog-lint file1.v file2.sv ...`` 
* The user will be able to add in apio.io a list of additional verible flags that will be appended to the command.  Those flags can include for example --rules with rules customizations or -rules-config which points to user's file in the project with the linter configuration.  See https://chipsalliance.github.io/verible/verilog_lint.html for verible's flags.

@Obijuan, @cavearr, any thoughts?

I plan to evaluate the plans using  'apio raw -- verible-verilog-format ... ' and go from there.

cavearr, 2024-12-20:
@zapta, I really like the plan, in fact I will add verible support for the icestudio editor, which could be real-time verification.
The idea is that it uses the same configuration rules or lint config that celery uses, I will wait for your progress to replicate it and use it.

about that:

"The user may delete only part of the project, for example by specifying one or more files or by specifying a module (?)."

I think it should be possible to check each individual file in the project or the entire project; setting granularity at the API level for concise modules could require a lot of work and i think is not necessary.

cavearr, 2024-12-20:
Hi team! @zapta, I'm thinking today and it's just a thought that you have to take calmly and meditate on it

With verible we have a very good linter, but I think it wouldn't be bad to have a new "verify" in addition to lint, and this new verify command should be the current lint with verilator.

Why? Because in many cases the linter can check that the design is fine (that the code is syntactically correct) but deep down the design may have errors that the verilator can detect because it "compiles" the verilog and obtains more information than the linter.

Why is it interesting? Because with large designs we can have different levels of checking that imply long waits and with this additional command we would facilitate the process, I think it could be something like this:

1) Level 0 -> linter with verible, very fast, you don't wait to detect errors.
2) Level 1 -> verify, slower than the linter but analyzes deeper and is still reasonably fast.
3) Level 2 -> Build -> yosys + npr, slower and for large designs it could be very slow, the user could have to wait many minutes to see the result.

If we only have level 0 (apio lint) and level 2 (apio build) the user may have to wait many minutes to have the result and be able to see errors if there are any

I propose:

1) apio  lint -> verible
2) apio verify-> verilator (current apio lint with verilator)
3) apio build (as now...)

I'm thinking about it and let me know!

zapta, 2024-12-20:
Hi Carlos,

Thanks for the feedback.  How about this plan, we will have in the next
release a single linter (which you can choose if it will be 'fast' or
'deep'), and will see if there is a significant user demand for a faster or
deeper linter.

My understanding is that apio is intended to be "lean and mean" and adding
a second linter may be a premature optimization, especially if one
considers that it's available via the ``apio raw`` command.

Will this work for you?

cavearr, 2024-12-20:
Sounds good to me, let's go for it!

zapta, 2024-12-20:
Which tool do you choose for 'apio lint', verible or verilator?

cavearr, 2024-12-20:
If we adjuts us to the term "lint" the correct one will be verible, but if in this release we only take one, i'll choose  verilator.

why? because verilator detect errores more complex than verible, and users that use apio with visual studio code for example, could install verible as visual studio plugin and this code will be linting by verible in realtime. 

The step "apio lint" should give a more deeper error detection than only verible.

Sorry for going from one side to another, is the  good things of discussing (thanks a lot @zapta) at the end i think we could have the best  decision.

zapta, 2024-12-20:
Let's see if I got it correctly, we will have a single command called
'verify' that uses verilator. Is it so?

cavearr, 2024-12-20:
this should be the ideal

zapta, 2024-12-20:
I will rename 'lint' to 'verify'.

cavearr, 2024-12-20:
Ok! then tell me when you finish to adapt in Icestudio please.

closed 2024-12-20 by zapta

zapta, 2024-12-20:
@cavearr, I started to renamed 'lint' to 'verify' and realized that it doesn't make sense. We run verilator with the flag ``--lint-only`` and generate a configuration file that controls linting (see below).  Do you want to reconsider your decision to rename it?

```
        "`verilator_config\n"
        f'lint_off -rule COMBDLY      -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
        f'lint_off -rule WIDTHEXPAND  -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
        f'lint_off -rule PINMISSING   -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
        f'lint_off -rule ASSIGNIN     -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
        f'lint_off -rule WIDTHTRUNC   -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
        f'lint_off -rule INITIALDLY   -file "{vlt_path(YOSYS_LIB_DIR)}/*"\n'
```