FPGAwars/apio #508: [For discussion] Use deeper apio commands.

zapta, 2024-12-23:
Apio command are limited to two levels and use options as the third levels of operation selection. E.g.

```
apio packages --install ...
apio packages --uninstall ...
apio packages --list ...
```

This proposal is to use as many command levels as intuitivly needed, similar to platformio.  E.g.

```
apio packages install ...
apio packages uninstall ...
apio packages list ...
```

Platform io:
[image]

The motivation is CLI clarity, separating between selecting the operation and specifying variations via options. My understanding is that python-click supports it natively.

@Obijuan, @cavearr, any thoughts?

cavearr, 2024-12-23:
I use platformio a lot and i think they fight with this kind of decisions before us, and for me is a clean solution.

My vote is aligned with clone plantformio philosophy for apio.

Thanks @zapta !

zapta, 2024-12-25:
Hi @cavearr, ok, I will give it a try. I will first convert the 'apio packages' command and we will see how it goes.

```
$ apio packages -h
Usage: apio packages [OPTIONS] COMMAND [ARGS]...

  'apio packages' is a group of commands that manage the apio installed
  packages. These are not python packages but apio specific packages that
  contain various tools and data that are necessary for the operation of apio.
  These packages are installed after the installation of the apio python
  package using the command 'apio packages install'. Note that the list of
  available packages depends on the operatingsystem you use as some require
  more packages than others.

  See below the list of commands in the 'apio packages' group.

Options:
  -h, --help  Show this message and exit.

Commands:
  apio packages install    Install apio packages.
  apio packages uninstall  Uninstall apio packages.
  apio packages list       List apio packages.
  apio packages fix        Fix broken apio packages.
```

and then 

```
$ apio packages install -h
Usage: apio packages install [OPTIONS] [PACKAGES]...

  The command 'apio packages install' installs apio packages which are require
  for the operation of apio on your system. Note that the list of available
  packages depends on the operatingsystem you use as some require more
  packages than others.

  Examples:
    apio packages install                   # Install all missing packages.
    apio packages install --force           # Re/install all missing packages.
    apio packages install oss-cad-suite     # Install a specific package.
    apio packages install examples@0.0.32   # Install a specific version.

  Adding the option --force to forces the reinstallation of existing packages,
  otherwise, packages that are already installed correctly are left with no
  change.

Options:
  -f, --force    Force installation.
  -v, --verbose  Show detailed output.
  -h, --help     Show this message and exit.
```

zapta, 2024-12-25:
Hi @cavearr, I split the first command, 'apio packages' into 4 sub commands. It's in this long pending PR. Can you accept it?

https://github.com/FPGAwars/apio/pull/507

Will do the same for a few other commands.

zapta, 2024-12-27:
Hi @cavearr, I completed splitting the multi-operations commands into subcommands.  You can see a report about the commands hierarchy and their help text here https://github.com/FPGAwars/apio/blob/develop/COMMANDS.md

Most likely you will need to adapt icestudio. If you have any questions about the adaptations or the new commands please let me know.

zapta, 2024-12-28:
Implemented, commands are here https://github.com/FPGAwars/apio/blob/develop/COMMANDS.md

Closing.

closed 2024-12-28 by zapta