FPGAwars/apio #509: gtkwave on macOS 12.7.6

daveti, 2024-12-25:
Command option --rcvar "splash_disable on" --rcvar "do_initial_zoom_fit 1" causes no response from gtkwave on my macOS 12.7.6. In fact, removing this option works like a charm. Not sure if this happens for newer macOS, but might be worth adding an OS and version check to workaround this issue.

zapta, 2024-12-25:
Hi @daveti,  what do you mean by 'causes no response', does gtkwave hang?

With the dev version, we use gtkwave 3.4.0 and I am testing it on macOS 14.5 and it works as expected,  What gtkwave version do you have?

A couple of suggestions for you in an arbitrary order
1. You can ask the gtkwave team here https://github.com/gtkwave/gtkwave/issues
2. You can try using the apio dev version. It's not fully tested but it depends how criticlal your application is (I use it for my own projects)
3. You can patch the command in your installed apio. Find the location of the apio python package ('pip show apio' will give you a hint),  find the SConstruct file for the architecture you use (e.g apio/resources/ice40/Sconstruct) and patch the gtkwave template there.

daveti, 2024-12-25:
Thank you for the quick response. Yup, my gtkwave hung there forever and could not even get started. I'm using gtkwave 3.3. Anyway, I can patch apio easily to work around this problem but figured it might be a good idea to document this issue, which might be related to my specific configuration here, e.g., gtkwave 3.3 + macOS 12.7.6.

Thanks again for the prompt feedback. Merry Xmas.

closed 2024-12-25 by daveti