FPGAwars/apio #515: [For discussion] Disallow unicode characters in apio home dir?

zapta, 2024-12-29:
TL;DR, proposal to drop support for unicode chars in apio home dir (default ``~.apio``).  (No change regarding apio project home dir).

For apio project dir we allow spaces and unicode characters, no problem here. For apio home dir (default ``~/.apio``), we allow unicode characters with no spaces but this create a problem with the gowin bitstream generator on windows (see test failure below).

I am proposing to drop the support for unicode characters in the apio home path, requiring a 'clean' ascii path.

Rationale:
1. Most users will not be affected.
2. Users with unicode user names can use $APIO_HOME_DIR to override.

@Obijuan , @cavearr, any thoughts?

-------------

Below is Gowin ``apio build`` failure when running the test on windows. Notice how the ``Temp\fuññy-7264an2r\shared-apio`` in the directory path is interpreted by the gowin tool as ``\Temp\\fu\udcf1\udcf1y-7264an2r\\shared-apio``, which leads to a ``file not found`` error. If I change ``fuññy`` to ``funny``, the test passes successfully. BTW, the problem exists on Windows but not on mac OSX.
```
--> apio sandbox:
       sandbox dir       : C:\Users\user\AppData\Local\Temp\fuññy-mhzu_5b5
       apio proj dir     : C:\Users\user\AppData\Local\Temp\fuññy-mhzu_5b5\ proj
       apio home dir     : C:\Users\user\AppData\Local\Temp\fuññy-7264an2r\shared-apio

...

E         yosys -p "synth_gowin -top blinky -json _build\hardware.json" -q blinky.v
E         nextpnr-himbaechel --device GW1NSR-LV4CQN48PC7/I6 --json _build\hardware.json --write _build\hardware.pnr.json --report _build\hardware.pnr --vopt family=gw1ns-4 --vopt cst=blinky.cst -q
E         gowin_pack -d GW1NS-4 -o _build\hardware.fs _build\hardware.pnr.json
E         C:\Users\user\AppData\Local\Temp\fu\udcf1\udcf1y-7264an2r\shared-apio\packages\tools-oss-cad-suite\bin\..\lib\python3.exe: can't open file 'C:\\Users\\user\\AppData\\Local\\Temp\\fu\udcf1\udcf1y-7264an2r\\shared-apio\\packages\\tools-oss-cad-suite\\bin\\gowin_pack-script.py': [Errno 2] No such file or directory
E         scons: *** [_build\hardware.fs] Error 2
``

zapta, 2025-01-03:
Ok, I restricted the apio home dir to ascii characters in the range [33 to 127] (no spaces and no accents).  If an invalid character is detected, apio gives this error message:

```
Error: The apio home dir contains a non supported character [ñ].
Apio home dir: '/Users/user/.apio-ñ'
Only the ASCII cractrs from 33 to 127 are supported. You can use the
system env var 'APIO_HOME_DIR' to set a different apio home dir.
```

No change in the apio project dir which can contain spaces and non ascii chars.

zapta, 2025-01-03:
Implemented a restriction to ascii chars [33, 127].  Closing.

closed 2025-01-03 by zapta

cavearr, 2025-01-03:
Hi @zapta this could be a problem for icestudio windows users , is really difficult support it?

cavearr, 2025-01-04:
Hi @zapta apio home dir in Icestudio is under Icestudio home dir, don't worry i'll  advice for users with weird characters and spaces in their paths that is not allowed, Gowin support is important in this new release.

Thanks!

zapta, 2025-01-04:
Thanks @cavearr.

There is another issue with gowin on mac OSX, the apio home directory must be in a partition that is not case sensitive (this is the default on mac OSX.  On linux partitions are case sensitive so it may be a problem (?). Can you test on linux?

Details here https://github.com/FPGAwars/apio/issues/521