FPGAwars/apio #522: [For discssion] Apio should require APIO_HOME_DIR, if set,  to have an absolute path

zapta, 2025-01-04:
This is a proposal to require the value of APIO_HOME_DIR, if used, to be an absolute path. 

Motivation: if  ``APIO_HOME_DIR=foo`` it's easy to create unintentionally a new home with packages in each directory where we run ``apio packages install`` or even just ``apio examples list`` (which also loads the packages if they don't exist).

For icestudio, this means that the path to the home directory under the project directory needs to be converted to an absolute path.

@cavearr, is this ok?

zapta, 2025-01-05:
Hi @cavearr, icestudio can set APIO_HOME_DIR with an absolute path, right?  (e.g. under user home directory).

zapta, 2025-01-09:
Implemented. Closing. 

I presume that it shouldn't be too difficult for icestudio to use an absolute path if it doesn't do it already.

closed 2025-01-09 by zapta

cavearr, 2025-01-09:
Sorry for my delay @zapta!, Icestudio use absolute path from the begining , this is ok!