FPGAwars/apio #523: [For discussion] What Python versions should apio support?

zapta, 2025-01-05:
Currently (Jan 20225) apio is tested with python versions **3.9/10/11/12/13**. This issue is a proposal to limit the testing to the last three python versions, in this case **3.11/12/13,** and possibly a release candidate for the upcoming 3.14.

A summary of python's versions
https://devguide.python.org/versions/

Motivation:
* Dropping old version allows to use newer features of the language. For example, we still can't use ``@override`` in apio.
* The more versions we target, the more work it is to make all of them happy. 
* The test time is proportional to the number of versions we test.

Dropping versions 3.9, 3.10 from the test also means that apio doesnt support them and most likely will fail with those versions.

@Obijuan, @cavearr, any thoughts?

The change will be applied here https://github.com/FPGAwars/apio/blob/86fac3de30c0ee276b088d088d180c92329a0ebd/tox.ini#L42

zapta, 2025-02-05:
@cavearr, what do you think of removing the apio support for python ``3.9``, ``3.10``, leaving only ``3.11``, ``3.12``, ``3.13``?

This allow to use newer language features such as ``@override`` and reduce the testing time?

zapta, 2025-05-07:
Since there are no opinions, I will go with supporting the the latest three python stable minor versions. E.g. 3.11, 3.12, 3.13.  This will affect pyproject.py, tox,ini, and the github workflows.

Installer building workflows will always generate the latest of the three.

zapta, 2025-05-07:
TODO:  change support to 3.11, 3.12, 3.13

zapta, 2025-05-07:
Hi @cavearr, any objection for apio moving the min python version from 3.9 to 3.11 ?  (supporting 3.11, 3.12, 3.13).

cavearr, 2025-05-07:
For me is ok! @zapta

zapta, 2025-05-08:
Dropped python 3.9, 3.10.  Apio now supports 3.11, 3.12, 3.13.   (3.13 is currently the latest stable python version).

Closing.

closed 2025-05-08 by zapta