FPGAwars/apio #529: Can we avoid the special case for tinyprog on darwin?

zapta, 2025-01-10:
@cavearr, this is a question for you.

The general command line for the tinyprog progammer that apio uses is 

``tinyprog --pyserial -c ${SERIAL_PORT} --program $SOURCE``

However there is a special hardcoded case for tinyprog on darwin which sets the command

``tinyprog --libusb --program $SOURCE``

Is this special case necessary? Can we merge the cases together into a single command line?

cavearr, 2025-01-10:
i'll try it . In osx always i use --libusb, the other command line i'm not use never.
Tomorrow afternoon i'll play with it in different OSs and try to find if is it possible reduce to one command.

Thanks @zapta!

zapta, 2025-01-10:
What board do you use with tinyprog?

I look for TinyFPGA BX and EX but couldn't find any for sale

https://tinyfpga.com/

zapta, 2025-01-17:
Hi @cavearr, can we remove the special case of tinyprog on darwin that sets the command line to ``tinyprog --libusb --program $SOURCE`` instead of the command ``tinyprog --pyserial -c ${SERIAL_PORT} --program $SOURCE`` that is used on the other platforms.

zapta, 2025-02-05:
Hi @cavearr , can we remove this tinyprog special case?  (more details in the other posts in this issue)

https://github.com/FPGAwars/apio/blob/cd6821a968e14891c6b47e6d365051fbeef5e333/apio/managers/programmers.py#L61

cavearr, 2025-02-05:
yes @zapta remove it, i'll work around tinyfpga the next week, i have been planned a deep work on it to fix the programmers , i'm telling you around this the next week because i think we need to use our own programmer to have all tinybx models working.

zapta, 2025-02-05:
Thanks @cavearr, I will remove it.

zapta, 2025-02-05:
Removed. Closing.

closed 2025-02-05 by zapta