FPGAwars/apio #531: tinyprog for tinyfpga-bx

cavearr, 2025-01-10:
Hi@zapta! I'm starting this thread to group the entire integration around tinyprog because it could take a long time and cause different problems, all related to each other, this way I think it will be easier to keep track of.

In this thread I will explain the evolution of our own patched version that works with the new oss cad suite and the new boards.

On the other hand, I'll start by asking you if you could check if Apio uses oss cad suite tinyprog or one downloaded from pip and installed on the system.

I haven't been able to verify it yet because I'm working on our tinyprog branch and it's giving me a lot of work (in parallel I'm releasing some progress tests in Icestudio to the community) and if you could help me, verifying this in Apio could be very useful.

I'm confused because if you try to "load" a tinyfpga-bx project with Apio (try it even if you don't have the board), Apio tells you when you call **apio upload** that you don't have tinyprog and that you need to install it from pip.   Because of this, I think Apio is not using oss-cad-suite tinyprog, but I could be wrong. Could you check it?

zapta, 2025-01-10:
Hi @cavearr, I will check it later today and will update here.  I see a few references in apio to the pip tinyprog package and am not sure why, if we already have it in oss-cad-suite. 

My guess is that apio looks for tinyprog before it set the yosys env vars and thus doesn't find it.  Try to run the tinyprog command with apio raw (which sets the env).

```
pip uninstall tinyprog         # Make sure the pip tinyprog doesn't exist
which tinyprog                 # Should not find tinypgor.
apio raw -- which tinyprog     # Should find tinyprog in oss-cad-suite
apio raw -- tinyprog ... <enter here the tinyprog args>..
```

cavearr, 2025-01-10:
It could be because in previous versions of celery, oss cad suite would not include tinyprog and Apio has not evolved with this tool.

Apio always looks for tinyprog in pip, I think we should remove these dependencies from pip and use oss-cad-suite.

TinyFpga is "abandoned" but many users have this card and there are several tutorials, videos and online workshops with Apio and the tinyfpga. It is important that we make it work.

zapta, 2025-01-10:
@cavearr, try to remove in apio/resources/programmers.json the tinyprog programmer reference to the pip package

https://github.com/FPGAwars/apio/blob/54bbe92f3d2b9113207d16314d67dc1100f79cfe/apio/resources/programmers.json#L46

```

# Before
  "tinyprog": {
    "command": "tinyprog",
    "args": "--pyserial -c ${SERIAL_PORT} --program",
    "pip_packages": [ "tinyprog" ]
  },

# After
  "tinyprog": {
    "command": "tinyprog",
    "args": "--pyserial -c ${SERIAL_PORT} --program",
    "pip_packages": [  ]
  },

```

zapta, 2025-01-10:
Sounds good. I will remove the tinyprog dependency on pip package.

cavearr, 2025-01-10:
Thanks @zapta , now i can't try it, if you can't i'll try tomorrow .

Thanks again!

zapta, 2025-01-10:
@cavearr, you can try it on your side by removing the pip package reference from the prorgammer definition in apio/resources/programmers.json 

https://github.com/FPGAwars/apio/issues/531#issuecomment-2584187443

I will do it separately on my side, test and send a PR.

cavearr, 2025-01-17:
> @cavearr, the tinyprog pip packages was removed and now tinyprog is executed from the oss bin. Can we close this issue?
>
> $ apio raw -- which tinyprog
> /Users/user/.apio/packages/oss-cad-suite/bin/tinyprog

Yes!! Thanks!!

closed 2025-01-18 by zapta