FPGAwars/apio #535: Board 'odt-rpga-feather' has an unknown programmer type "mcu".

zapta, 2025-01-11:
@Obijuan , @cavearr 

The board definition below has the programmer type "mcu" but there is not such programmer in the programmers.json file.  Is it broken? Any suggestions on how to fix it?

https://github.com/FPGAwars/apio/blob/6cd40d3068ee2164af3da4f3c295c862ac66c04d/apio/resources/boards.json#L665

```
  "odt-rpga-feather": {
    "legacy_name": "ODT_RPGA_Feather",
    "description": "ODT_RPGA_Feather",
    "fpga": "iCE40-U4K-SG48",
    "programmer": {
      "type": "mcu"
    }
  },
```

zapta, 2025-01-13:
@Obijuan, @cavearr, do you know what this 'mcu' mean?  Is this board useable with apio?

cavearr, 2025-01-13:
i don't know this board

zapta, 2025-01-13:
@cavearr, it was added in the commit below. Even at that time, there was no programmer called 'mcu' so it seems that that board was broken from the begining.

Ok to delete?

https://github.com/FPGAwars/apio/commit/b427656afa47ecfdacaeeb26d5fb82df54cc2b45

cavearr, 2025-01-13:
> b427656

For me is ok, but wait for @obijuan responde.

zapta, 2025-01-13:
Hellow @skerr92, can you please shed a light on this commit https://github.com/FPGAwars/apio/commit/b427656afa47ecfdacaeeb26d5fb82df54cc2b45 ?

The reason we ask is that we are revisiting the apio boards definitions and board ``odt-rpga-feather`` refers to the apio programmer id ``mcu`` which doesn't see to exist in apio's programmers definitions:  https://github.com/FPGAwars/apio/blob/develop/apio/resources/programmers.json

Was it ever run successfully with apio?

skerr92, 2025-01-13:
The goal with MCU was that the FPGA on RPGA Feather is programmed by an RP2040. The primary method intended for loading the FPGA was through a library used with CircuitPython. The user would end up needing to copy the bin to the CIRCUITPY drive. I was hoping to revisit this and add a check for CIRCUITPY drive and copy over automatically. Alternatively, transforming the bin to a C header to use with Arduino.

zapta, 2025-01-13:
Thanks @skerr92.  The programmer type in apio's board definition is a key reference to programmers.json when the programmer commands that apio issue are defined.

https://github.com/FPGAwars/apio/blob/develop/apio/resources/programmers.json

1. Do you know if there is a programmer command for that board?  E.g. a python pip package that installed a binary that apio can call to program the board?

2. That board definition never upload successfully with apio, right?  (we try to decide how to handle it).

cavearr, 2025-01-13:
@zapta , @skerr92 

if mcu is in development, the best solution now is used the local boards.json and programmers.json until the development will be finished or almost with something working. In this moment the mcu will be part of apio programmers.

Is this the case? if is it, the best solution is remove mcu from apio and return back when works. now is not neccesary inetgrate into apio core  like before this new version.

zapta, 2025-01-13:
@cavearr, removing it for now until it will be resolved sounds reasonable.

@skerr92, if I understand correctly, the approach you propose is not finding a way to program that board with apio/icestudio but to have a graceful way to mark at board as  'non programmable with apio/icestudio'.  Is it so?

skerr92, 2025-01-13:
Yeah, that's correct. A graceful failure/notification so it's generally still usable.

zapta, 2025-01-14:
What is the use case?  Apio will generate the bitstream and will not
upload. What will happen next in this scenario?

If the goal is just to learn verilog programming without using a board, the
user can pick any board id they want and just not upload, only linting,
simulating, etc.

zapta, 2025-01-14:
Thanks. I created an issue for it here https://github.com/FPGAwars/apio/issues/547

For now we will comment out ``odt-rpga-feather`` because that feature is not supported yet.

zapta, 2025-01-16:
Board removed. Added an issue with a feature request for the 'no programming programmer'. Closing.

closed 2025-01-16 by zapta