FPGAwars/apio #536: Programmers definitions 'openfpgaloader', 'ujprog' are not used.

zapta, 2025-01-11:
The two programmers 'openfpgaloader', 'ujprog' are defined but not used. 

@juan, @cavearr, OK to delete?  (we can also add back when needed).

Motivation: I am adding an automatic test that validates the boards, fpgas, and programmers and this breaks the test.

https://github.com/FPGAwars/apio/blob/develop/apio/resources/programmers.json

Obijuan, 2025-01-13:
Please, do not remove the programmers by now. Some boards can be programmed using different programmers. Buy default one is used, but the others can also be used. It is good for testing. For example, the Alhambra-II board uses iceprog, but it can also use openFPGALoader, which is faster (I am planning to switch to it)

Before removing things I think it is better to mark them as deprecated, and give more time to other uses to test. The current apio development branch has too many changes. I fear that when releasing it, many projects will be broken

cavearr, 2025-01-13:
@Obijuan , in Icestudio i'm evolving the compiler to add verilator exceptions for some things that broken a lot of projects at verify process.

The big "projects conflicts" comes from **verify** with verilator, but this is good in medium term because the problems are really bad practice implementations. I'm requesting at fpgawars community to test with projects and send me each of this that verify fails to add exceptions or evolve the error handling.

Don't worry with that, i think we don't needed of release nothing that has not been tested for months and all of the problems will be solved before.

The migration to openfpgaloader i thinks is a very good idea, i use a lof and thinks is better than iceprog, if you need  help with this, tell us.

I'm ok with no delete this programmers, in fact we could do in near future that in apio.ini some boards could be configure tu use one or other, what do you think @zapta ? Setup a preferrer programmer for board, but do the capability of setup an alternative programmer per board, for example Alhambra-ii, default programmer iceprog but alternative could be openfpgaloader defined in apio.ini.

zapta, 2025-01-13:
Sounds good, I will leave it. The 'openfpgaloader' is in use now for the gowin/tang boards so `ujprog` is the only one that is unused.

https://github.com/zapta/apio_dev/blob/2cd73f2f33c6a0cae48776c2a41ffca6e1e34eaa/test/test_apio_context.py#L97

As for breaking existing project, they are supposed to break only were we had concuss decisions such as requiring apio.ini or dropping support for custom SConstruct.

```

$ apio boards
BOARD                              EXAMPLES   ARCH     PART NUMBER              SIZE   PROGRAMMER                    
alchitry-cu                           1       ice40    ICE40HX8K-CB132          8k     iceprog                       
alhambra-ii                           4       ice40    ICE40HX4K-TQ144          8k     iceprog                       
arice1                                        ice40    ICE40UP5K-SG48           5k     iceprog                       
blackice                              2       ice40    ICE40HX4K-TQ144          8k     blackiceprog                  
blackice-ii                                   ice40    ICE40HX4K-TQ144          8k     blackiceprog                  
blackice-mx                                   ice40    ICE40HX4K-TQ144          8k     blackiceprog                  
edu-ciaa-fpga                         4       ice40    ICE40HX4K-TQ144          8k     iceprog                       
fomu                                  2       ice40    ICE40UP5K-UWG30          5k     dfu                           
fpga101                                       ice40    ICE40UP5K-SG48           5k     iceprog                       
go-board                              3       ice40    ICE40HX1K-VQ100          1k     iceprog                       
ice40-hx1k-evb                        1       ice40    ICE40HX1K-VQ100          1k     iceprogduino                  
ice40-hx8k                            1       ice40    ICE40HX8K-CT256          8k     iceprog                       
ice40-hx8k-evb                        1       ice40    ICE40HX8K-CT256          8k     iceprogduino                  
ice40-ul1k-breakout                           ice40    ICE40UL1K-CM36A          1k     iceprog                       
ice40-up5k                            3       ice40    ICE40UP5K-SG48           5k     iceprog                       
iceblink40-hx1k                               ice40    ICE40HX1K-VQ100          1k     iceburn                       
icebreaker                            3       ice40    ICE40UP5K-SG48           5k     iceprog                       
icebreaker-bitsy0                             ice40    ICE40UP5K-SG48           5k     dfu                           
icebreaker-bitsy1                             ice40    ICE40UP5K-SG48           5k     dfu                           
icefun                                        ice40    ICE40HX8K-CB132          8k     icefunprog                    
icestick                              2       ice40    ICE40HX1K-TQ144          1k     iceprog                       
icesugar-1-5                          1       ice40    ICE40UP5K-SG48           5k     icesprog                      
icesugar-nano                                 ice40    ICE40LP1K-CM36           1k     icesprog                      
icewerx                               2       ice40    ICE40HX8K-CB132          8k     icefunprog                    
icezum                                5       ice40    ICE40HX1K-TQ144          1k     iceprog                       
kefir                                 2       ice40    ICE40HX4K-TQ144          8k     iceprog                       
odt-icyblue-feather                           ice40    ICE40U4K-SG48            4k     icesprog                      
odt-rpga-feather                              ice40    ICE40U4K-SG48            4k     mcu                           
ok-ice40pro                                   ice40    ICE40UP5K-SG48           5k     iceprog                       
pi-sicle                                      ice40    ICE40HX4K-TQ144          8k     pi-sicle-loader               
pico-ice                                      ice40    ICE40UP5K-SG48           5k     dfu                           
thetamachines-eth4k                           ice40    ICE40HX4K-TQ144          8k     iceprog                       
tinyfpga-b2                           2       ice40    ICE40LP8K-CM81           8k     tinyfpgab                     
tinyfpga-bx                           4       ice40    ICE40LP8K-CM81           8k     tinyprog                      
upduino                                       ice40    ICE40UP5K-SG48           5k     iceprog                       
upduino2                                      ice40    ICE40UP5K-SG48           5k     iceprog                       
upduino21                                     ice40    ICE40UP5K-SG48           5k     iceprog                       
upduino3                                      ice40    ICE40UP5K-SG48           5k     iceprog                       
upduino31                             3       ice40    ICE40UP5K-SG48           5k     iceprog                       
butterstick-r10-2g-85k                        ecp5     LFE5UM5G-85F-8BG381C     85k    dfu                           
butterstick-r10-2g-85k-ft2232h                ecp5     LFE5UM5G-85F-8BG381C     85k    openfpgaloader-ft2232         
[... cut ...]
sipeed-tang-nano-1k                           gowin    GW1NZ-LV1QN48C6/I5       1k     openfpgaloader                
sipeed-tang-nano-20k                          gowin    GW2AR-LV18QN88C8/I7      20k    openfpgaloader                
sipeed-tang-nano-4k                   1       gowin    GW1NSR-LV4CQN48PC7/I6    4k     openfpgaloader                
sipeed-tang-nano-9k                   2       gowin    GW1NR-LV9QN88PC6/I5      9k     openfpgaloader                
Total of 80 boards
Run 'apio boards -v' for additional columns.
```

cavearr, 2025-01-13:
i like a lot all of the options, the board.json in the project folder for custom boards is very interesting, this is very aligned with a new feature to add boards that i'm developing at icestudio.

For me and for the momment with this should be enought, maybe in apio.ini needs and "args" parameter for the programmer.

zapta, 2025-01-13:
BTW, the programmers.json is also customizable and users can add them to
their projects as well.

https://github.com/FPGAwars/apio/blob/884eabd14858967295cb7ba538128914d6c68c8f/apio/apio_context.py#L185

zapta, 2025-01-13:
I created an issue to discuss the board and programmer replication issue.  https://github.com/FPGAwars/apio/issues/new

Closing this one. The unused programmer will say.

zapta, 2025-01-18:
@Obijuan, the ``ujprog`` programmer definition is not only unused by also doesn't have a matching binary, so broken.   Ok to delete?

(There is another programmer definition called ``fujprog`` and it seems to be ok).

```
  "ujprog": {
    "command": "ujprog",
    "args": "-q -j SRAM"
  },
```

```
/projects/apio-dev/repo$ apio raw -- which ujprog
/projects/apio-dev/repo$
```