FPGAwars/apio #537: Fix icestudio test-examples to pass 'apio lint'.

zapta, 2025-01-11:
@cavearr, can you look into this?

the apio/test-examples dir contains a few tens of project that were generated with icestudio and they fails 'apio lint'.  This issue is to fix them, for example, by generating again their apio projects with the latest icestudio. 

Example 1
```
/projects/apio-dev/repo/test-examples/ice40/alhambra-ii$ apio lint -p icestudio-ledon/
Setting the envinronment.
[Sat Jan 11 12:37:01 2025] Processing alhambra-ii
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Scanning for issues.
Testbench ledon_tb.v
verilator_bin --lint-only --quiet --bbox-unsup --timing -Wno-TIMESCALEMOD -Wno-MULTITOP --top-module main -DNO_ICE40_DEFAULT_ASSIGNMENTS -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40" _build/hardware.vlt "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40/cells_sim.v" main.v ledon_tb.v
%Warning-ASCRANGE: main.v:8:9: Ascending bit range vector: left < right of bit range: [0:6]
                             : ... note: In instance 'main'
    8 |  output [0:6] vinit
      |         ^
                   ... For warning description see https://verilator.org/warn/ASCRANGE?v=5.031
                   ... Use "/* verilator lint_off ASCRANGE */" and lint_on around source to disable this message.
scons: *** [_build/hardware] Error 1
============================================================================= [ ERROR ] Took 0.23 seconds =============================================================================
```

Example 2
```
/projects/apio-dev/repo/test-examples/ice40/alhambra-ii$ apio lint -p icestudio-riscv-stop-watch/
Setting the envinronment.
[Sat Jan 11 12:37:07 2025] Processing alhambra-ii
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Scanning for issues.
Testbench 03-riscv-stop-watch_tb.v
verilator_bin --lint-only --quiet --bbox-unsup --timing -Wno-TIMESCALEMOD -Wno-MULTITOP --top-module main -DNO_ICE40_DEFAULT_ASSIGNMENTS -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40" _build/hardware.vlt "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40/cells_sim.v" main.v 03-riscv-stop-watch_tb.v
%Warning-PINMISSING: main.v:85:10: Cell has missing pin: 'v8d2eee'
   85 |  vf1cffe v468719 (
      |          ^~~~~~~
                     main.v:150:9: ... Location of port declaration
  150 |  output v8d2eee
      |         ^~~~~~~
                     ... For warning description see https://verilator.org/warn/PINMISSING?v=5.031
                     ... Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: main.v:125:10: Cell has missing pin: 'vb385cd'
  125 |  v04e061 vb15d38 (
      |          ^~~~~~~
                     main.v:2583:15: ... Location of port declaration
 2583 |  output [4:0] vb385cd,
      |               ^~~~~~~
%Warning-PINMISSING: main.v:125:10: Cell has missing pin: 'vd9f5b6'
  125 |  v04e061 vb15d38 (
      |          ^~~~~~~
                     main.v:2584:9: ... Location of port declaration
 2584 |  output vd9f5b6
      |         ^~~~~~~
%Warning-PINMISSING: main.v:626:10: Cell has missing pin: 'v51fb1f'
  626 |  v794b6d va8ea8d (
      |          ^~~~~~~
                     main.v:782:16: ... Location of port declaration
  782 |  output [21:0] v51fb1f,
      |                ^~~~~~~
%Warning-PINMISSING: main.v:630:10: Cell has missing pin: 'v7e4f0f'
  630 |  vaaf5c4 ve8e400 (
      |          ^~~~~~~
                     main.v:817:9: ... Location of port declaration
  817 |  output v7e4f0f
      |         ^~~~~~~
%Warning-PINMISSING: main.v:635:10: Cell has missing pin: 'v712289'
  635 |  vaaf5c4 v677471 (
      |          ^~~~~~~
                     main.v:814:8: ... Location of port declaration
  814 |  input v712289,
      |        ^~~~~~~
[... 104 lines cut ...]
 2134 |  reg q = INI;
      |      ^
        main.v:2131:9: ... Location of original declaration
 2131 |  output q
      |         ^
%Error: main.v:3889:6: Duplicate declaration of signal: 'q'
 3889 |  reg q = INI;
      |      ^
        main.v:3886:9: ... Location of original declaration
 3886 |  output q
      |         ^
%Error: main.v:4089:6: Duplicate declaration of signal: 'q'
 4089 |  reg q = INI;
      |      ^
        main.v:4087:9: ... Location of original declaration
 4087 |  output q
      |         ^
%Error: main.v:4235:6: Duplicate declaration of signal: 'q'
 4235 |  reg q = INI;
      |      ^
        main.v:4233:9: ... Location of original declaration
 4233 |  output q
      |         ^
scons: *** [_build/hardware] Error 1
```

cavearr, 2025-01-11:
Hi @zapta, tomorrow i'll fix them, this is not a problem of Icestudio is for the verilog inside.

The examples are very old and yosys  get more stricted year by year and this examples contains errors.

Don't worry tomorros i'll take this task and fix it

zapta, 2025-01-11:
Thanks Carlos.  And feel free to delete examples from test-examples. I
think we have more than we need and the automatic tests use the examples
from 'apio examples' anyway.

cavearr, 2025-01-11:
i'm on it now, bug some of this have many problems, i'll try to fix as soon as possible

cavearr, 2025-01-12:
@zapta , finally i could work on it and all is fixed. You could try to lint ;)

zapta, 2025-01-12:
Thanks Carlos, I will git it a try

zapta, 2025-01-12:
Hi @cavearr, thanks for fixing it.  

1. Four examples lint with no problem. One still gives a lint error, I think because of a non ascii char in the var name.

```
+ apio lint -p icestudio-jumping-led
Setting the envinronment.
[Sat Jan 11 19:59:18 2025] Processing alhambra-ii
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Scanning for issues.
Testbench 02-jumping-LED_tb.v
Warning: [02-jumping-LED_tb.v] Using $dumpfile() in apio testbenches is not recomanded.
verilator_bin --lint-only --quiet --bbox-unsup --timing -Wno-TIMESCALEMOD -Wno-MULTITOP --top-module main -DNO_ICE40_DEFAULT_ASSIGNMENTS -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40" _build/hardware.vlt "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40/cells_sim.v" main.v 02-jumping-LED_tb.v
%Error: 02-jumping-LED_tb.v:29:9: syntax error, unexpected end of file, expecting ',' or ';'
   29 |  reg Bot  n;
      |         ^
%Error: Cannot continue
scons: *** [_build/hardware] Error 1
```

2. The test benches trigger the apio warning below. It is recomanded not to use $dumpfile() since apio already set the correct dump file output in the command line.   Can you remove it from the testbenchs icestudio generates?

```
Warning: [ledon_tb.v] Using $dumpfile() in apio testbenches is not recomanded.

```

zapta, 2025-01-12:
3. Also, the examples fail ``apio test``.  For example 
```
/projects/apio-dev/repo/test-examples/ice40/alhambra-ii/icestudio-leds-buttons$ apio test
Setting the envinronment.
[Sat Jan 11 20:09:03 2025] Processing alhambra-ii
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Scanning for issues.
Testbench 01-LEDs-buttons_tb.v
Warning: [01-LEDs-buttons_tb.v] Using $dumpfile() in apio testbenches is not recomanded.
iverilog -g2012 -o _build/01-LEDs-buttons_tb.out -DVCD_OUTPUT=_build/01-LEDs-buttons_tb -DNO_ICE40_DEFAULT_ASSIGNMENTS -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40" "/Users/user/.apio/packages/oss-cad-suite/share/yosys/ice40/cells_sim.v" main.v 01-LEDs-buttons_tb.v
01-LEDs-buttons_tb.v:25: error: 'constant_Constant' has already been declared in this scope.
01-LEDs-buttons_tb.v:24:      : It was declared here as a parameter.
01-LEDs-buttons_tb.v:29: error: 'Button' has already been declared in this scope.
01-LEDs-buttons_tb.v:28:      : It was declared here as a variable.
01-LEDs-buttons_tb.v:31: error: 'LED' has already been declared in this scope.
01-LEDs-buttons_tb.v:30:      : It was declared here as a net.
01-LEDs-buttons_tb.v:33: error: 'LED' has already been declared in this scope.
01-LEDs-buttons_tb.v:30:      : It was declared here as a net.
01-LEDs-buttons_tb.v:34: error: 'LED' has already been declared in this scope.
01-LEDs-buttons_tb.v:30:      : It was declared here as a net.
scons: *** [_build/01-LEDs-buttons_tb.out] Error 5
================================================================================= [ ERROR ] Took 0.22 seconds =================================================================================
/projects/apio-dev/repo/test-examples/ice40/alhambra-ii/icestudio-leds-buttons$ 
```

cavearr, 2025-01-12:
Don' worry @zapta , i'm review today, i don't review the testbenches , i only had been generated from Icestudio but yesterday  i didn't could test it.

Today i'm reviewing all of this stuff and in the way i refine Icestudio linter parser and improve the error management.

cavearr, 2025-01-12:
@zapta , it's done, examples fixed!

zapta, 2025-01-12:
Thanks Carlos!

zapta, 2025-01-13:
Hi @cavearr, try to run this bash script at the root of the apio repo. It will test all the icestudio projects in test-examples, and will stop on first error.

BTW, I think that we have too many icestudio examples there so please feel free to delete some, this is also a form of fixing. 

```
#!/bin/bash

# Exit on first error.
set -e

# Find all apio.ini in icestudio projects
projects=$(find test-examples | grep icestudio | grep apio.ini)

for proj in $projects;  do
    # Get the parent directory of apio.ini
    proj="$(dirname "$proj")"

    echo
    echo "--- PROJECT  $proj"
    echo

    # -- Go to the project's dir.
    cd $proj

    # -- Exceute apio commands in the project. They should succeeed.
    set -x
      apio build
      apio lint
      apio test
      apio graph
      apio report
      apio clean
    set +x
    
done
```

zapta, 2025-05-07:
To reproduce:

1. Change to ``scripts`` directory in the ``apio`` repo.
2. Run ``./check-icestudio-test-examples.sh``
3. The script run ``apio lint`` on each icestudio examples it finds in ``test-examples`` and stops on the first failure.

@cavearr, since these is verilog code that is generated automatically by icestudio, assigning it to you. 

Completion criteria: the scripts completes with no lint failures.

cavearr, 2025-05-07:
Ok @zapta ! i'm taking it as soon as possible in the next days.

zapta, 2025-06-21:
Closing. We now have automatic testing that use test projects from the examples package.

closed 2025-06-21 by zapta