FPGAwars/apio #545: Resolve unknown ICE40 part numbers ICE40U4K-UWG30 and  ICE40U4K-SG48

zapta, 2025-01-14:
These two part numbers in fpga.json are unknown.  Resolve them, verify their attributes and verify the match to the odt boards that use the second one (48).

They were added originally in https://github.com/FPGAwars/apio/commit/b427656afa47ecfdacaeeb26d5fb82df54cc2b45

Boards
https://github.com/Oak-Development-Technologies/IcyBlue
https://github.com/Oak-Development-Technologies/RPGA-Feather

fpgas.json:
```
  "ice40-u4k-uwg30": {
    "part_num": "ICE40U4K-UWG30",    
    "arch": "ice40",
    "type": "u4k",
    "size": "4k",
    "pack": "uwg30"
  },
  "ice40-u4k-sg48": {
    "part_num": "ICE40U4K-SG48",     
    "arch": "ice40",
    "type": "u4k",
    "size": "4k",
    "pack": "sg48"
  }
```

boards.json:
```
  "odt-icyblue-feather": {
    "legacy_name": "ODT_IcyBlue_Feather",
    "description": "ODT_IcyBlue_Feather",
    "fpga": "ice40-u4k-sg48",
    "programmer": {
      "type": "icesprog"
    },
    "usb": {
      "vid": "0403",
      "pid": "6014"
    }
  },
  "odt-rpga-feather": {
    "legacy_name": "ODT_RPGA_Feather",
    "description": "ODT_RPGA_Feather",
    "fpga": "ice40-u4k-sg48",
    "programmer": {
      "type": "mcu"
    }
  },
```

zapta, 2025-01-14:
Hello @skerr92, do you know what are the part numbers of the two fpgas that were added in the commit below

https://github.com/FPGAwars/apio/commit/b427656afa47ecfdacaeeb26d5fb82df54cc2b45

We are in the process of matching apio fpga definitions to actual datasheet and digikey manufacturer part numbers.

skerr92, 2025-01-14:
These match up to the ICE5LP4K parts. Same family as the ICE40UP5K (SG48 packages are pin compatible). I was following the naming that ICE Storm uses for their CLI tools.

zapta, 2025-01-14:
Can you identify them in the attached ordering information (extracted from the datasheet)?  For example, I don't see any with 30 pins.

[iCE40 Ultra Family Data Sheet.pdf](https://github.com/user-attachments/files/18414020/iCE40.Ultra.Family.Data.Sheet.pdf)

zapta, 2025-01-14:
The 48 pin fpga seems to be ICE5LP4K-SG48I and this is confirmed by the IcyBlue schematic:

https://github.com/Oak-Development-Technologies/IcyBlue/blob/main/Images/IcyBlue%20Schematic.png

The 30 pin fpga seems to not exist, and since it's not used by any board, safe to delete. 

Does this make sense?

skerr92, 2025-01-15:
30 pin should have been 20 pin (there was a CWG20 variant) but it would now be a 36 pin variant SWG36. I'm not certain that exists in Ice Storm yet though.

zapta, 2025-01-15:
Thanks Seth. I will fix it.

BTW, is there an authoritative list of the FPGA part numbers that are supported by the open source tools and the pnr options for each one of them?

https://github.com/YosysHQ/icestorm/commit/42be8d5bf0c2c6f3422b14cc4d264da7781da43b

skerr92, 2025-01-15:
I tend to lean on what's in this list here: https://github.com/YosysHQ/icestorm/blob/master/icefuzz/icecube.sh

zapta, 2025-01-15:
Thanks @skerr92!

zapta, 2025-01-16:
The ICE40U4K-SG48 was resolved to ICE5LP4K-SG48, based on the board's picture.  The ICE40U4K-UWG30 does not exist (and is not used by any board definition) and thus removed. 

Closing.

closed 2025-01-16 by zapta