FPGAwars/apio #546: What is the fpga attribute iddecode and do we need it?

zapta, 2025-01-14:
Below is a snippet from fpgas.jsonc.  One fpga has an idcode field and a similar one doesn't..

1. What does idcode do?
2. When do we need to specify it, if at all?

```
  "ecp5-lfe5u-12f-csfbga285": {
    "part_num": "LFE5U-12F-6MG285C",
    "arch": "ecp5",
    "size": "12k",
    "type": "12k",
    "pack": "CSFBGA285",
    "idcode": "0x21111043"
  },
  "ecp5-lfe5u-25f-cabga256": {
    "part_num": "LFE5U-25F-6BG256C",
    "arch": "ecp5",
    "size": "25k",
    "type": "25k",
    "pack": "CABGA256"
  },
```

zapta, 2025-01-14:
Hi @mmicko, can you shed a light on the 'idcode' attributes of a few ecp5 fpgas definitions in apio?  They seem to originate in an old commit of you.

Apio fpga definitions:
https://github.com/FPGAwars/apio/blob/b2bad1b46c4cf868f6fe17bf70d0be61450328c8/apio/resources/fpgas.json#L251

Apio ecp5 architecture plugin:
https://github.com/FPGAwars/apio/blob/b2bad1b46c4cf868f6fe17bf70d0be61450328c8/apio/scons/plugin_ecp5.py#L115

mmicko, 2025-01-14:
Hi @zapta yes it is leftover from past, was required to support 12k, by building for 25k and then manually updating idcode with ecppack, it should not be used anymore since there is now direct support.

zapta, 2025-01-14:
Thanks @mmicko!

zapta, 2025-01-14:
idcode removed. Closing.

closed 2025-01-14 by zapta