FPGAwars/apio #549: [Feature Request] Indicate to the user which drivers are necessary for their boards

zapta, 2025-01-17:
Apio has commands such as ``apio drivers ftdi install`` or ``apio drivers serial install`` but how should the user know which driver they should install for their board?

This request is for providing such indication, either using the existing data in apio resource files or by adding new data for this purpose.

zapta, 2025-01-22:
@Obijuan, @cavearr, we have ``apio drivers ftdi install`` and ``apio drivers serial install``.  How can the user know if they need to install a driver and if so, which of the two?

Any suggestions on how to make it obvious?

cavearr, 2025-01-22:
In icestudio is configured in the board definition, could be the same in apio?  and in this case make only apio drivers install?

Could be more consistent change the command by ``apio drivers install`` and by default use ftdi and support pass ftdi (default) or serial as optional , for example ``apio drivers install serial``

What do you think @zapta , @Obijuan ?

zapta, 2025-01-22:
@cavearr 

1. Let's say we know the board, the fpga, and the programmer information from boards.jsonc,  fpgas.jsonc, and programmers.jsonc respectivly, and lets say the an upload operation failed. What message can we give the user to help them install the correct driver if needed?

Should we have to the programmer definition information about the driver requirements?

https://github.com/FPGAwars/apio/blob/develop/apio/resources/programmers.jsonc

2. The current driver commands are 

```
apio drivers frdi install
apio drivers frdi uninstall
apio drivers frdi list

apio drivers serial install
apio drivers serial uninstall
apio drivers serial list

apio drivers lsusb
```

Do you propose to change to 

```
apio drivers install ftdi
apio drivers install serial

apio drivers uninstall ftdi
apio drivers uninstall serial

apio drivers list ftdi
apio drivers list serial
apio drivers list usb
```

zapta, 2025-01-22:
@cavearr, I don't know. Another option is to add to apio.ini an option to add flags to yosys synth.

zapta, 2025-01-22:
For item 2, I think I prefer your suggestions because it makes the existing apio ``apio drivers lsusb`` more consistent with the ftdi and serial listing ``apio drivers list usb``.

cavearr, 2025-01-22:
tell me a day to think in it , please

zapta, 2025-01-22:
Sounds good. Whenever it works for you.

zapta, 2025-01-22:
@cavearr, do you  think on number 1 or number 2?

For number 2, I think your suggestion for ``apio drivers install ftdi`` is better than the old way.

zapta, 2025-01-22:
@cavearr, I implemented your idea for the apio drivers command structure. I think it's cleaner this way. You may need to adapt icestudio if you use the drivers commands.

The new commands are:

```
apio drivers list ftdi
apio drivers list serial
apio drivers list usb

apio drivers install ftdi
apio drivers install serial

apio drivers uninstall ftdi
apio drivers uninstall serial
```

cavearr, 2025-01-22:
Sorry for my delay! dinner time! 😆 

I like this option, when you do a PR y addapt iceStudio don't worry for that, i'm doing a lot of changes and refactoring in Icestudio , the wip is for this 😄 

I'm working and doing test around folder structure, tomorrow i'll post you around that.

Great work @zapta !

zapta, 2025-01-23:
On my darwin-arm64 machine, ``apio upload`` with alhambra-ii works regardless if I install or uninstall the ftdi and serial drivers.

cavearr, 2025-01-23:
Maybe in osx could be and old feature for old intel machines. This afternoon i'm reviewing the code to clear it.