FPGAwars/apio #552: how to allow yosys synth-ice40 -dsp switch

iamhuzhe-nia, 2025-01-21:
Currently no multiplier (DSP) resource is used with "apio build -v" command, on ice40 project.

It seems that with "-dsp" switch, it should solve the issue.

But where, how should that switch be invoked?

zapta, 2025-01-22:
Thanks for the report @iamhuzhe-nia.

1. To what binary did you add that option?  (yosys, nextpnr, etc).

2. Is it reasonable to have it specified all the time or do you expect that some users will want it on and some off?

iamhuzhe-nia, 2025-01-22:
Here is the line I modified in SConstruct file

original
 `    action='yosys -p \"synth_ice40 {0} -json $TARGET\" {1} $SOURCES'.format(
`
modified
 `    action='yosys -p \"synth_ice40 {0} -dsp -json $TARGET\" {1} $SOURCES'.format(
`

Not an expert on yosys, here is where I get my information about multiplier https://projectf.io/posts/multiplication-fpga-dsps/

So I don't understand if "-dsp" switch isn't on, how does yosys "synthesize" a correct multiplier using LUTs only.

cavearr, 2025-01-22:
Hello @zapta! I think this argument is important and should be able to be configured via celery.ini because in most cases even if the fpga has dsps the developer may not want to use it for some reason.

For me, this powerful argument should be optional and disabled by default.

zapta, 2025-01-22:
@cavearr, something like this?

```
[env]
board = alhambra-ii
top-module = main

# Extra args to yosys synth
yosys-synth-extra-options = -dpi

# Style preferences for 'apio format'.
format-verible-options =
    --column_limit=80
    --indentation_spaces=4

```

cavearr, 2025-01-22:
i like a lot! be careful! -dsp not -dpi! 😄

zapta, 2025-01-22:
Yes, -dsp.  ;)

I will implement it.

zapta, 2025-01-22:
@iamhuzhe-nia, can you post here a zip file with a small apio project that will benefit from the -dsp?  We will use it for testing and may also add it to the apio examples. The smaller and simpler, the better.

zapta, 2025-01-23:
I implemented a small apio (dev) example that demonstrates the -dsp feature.  https://github.com/zapta/apio_dev/tree/develop/test-examples/ice40/fomu/dsp  

You can see below how the synthesis trades differently ICESTORM_DSP and ICESTORM_LC cells and how it affects the design speed.

Without ``yosys-synth-extra-options = -dsp``:

```
$ apio report
Setting the envinronment.
[Wed Jan 22 22:07:20 2025] Processing fomu
---------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top top -json _build/hardware.json" -q blink.v
nextpnr-ice40 --up5k --package uwg30 --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf pinout.pcf -q
Formatting pnr report.

UTILIZATION:
        ICESTORM_DSP:     0     8     0%
      ICESTORM_HFOSC:     0     1     0%
         ICESTORM_LC:  1131  5280    21%
      ICESTORM_LFOSC:     0     1     0%
        ICESTORM_PLL:     0     1     0%
        ICESTORM_RAM:     0    30     0%
      ICESTORM_SPRAM:     0     4     0%
              IO_I3C:     0     2     0%
               SB_GB:     1     8    12%
              SB_I2C:     0     2     0%
               SB_IO:     6    96     6%
         SB_LEDDA_IP:     0     1     0%
         SB_RGBA_DRV:     0     1     0%
              SB_SPI:     0     2     0%
         SB_WARMBOOT:     0     1     0%

CLOCKS:
                clki:   20.47 Mhz max

```

With ``yosys-synth-extra-options = -dsp``:

```
$ apio report
Setting the envinronment.
[Wed Jan 22 22:06:31 2025] Processing fomu
---------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top top -dsp -json _build/hardware.json" -q blink.v
nextpnr-ice40 --up5k --package uwg30 --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf pinout.pcf -q
Formatting pnr report.

UTILIZATION:
        ICESTORM_DSP:     3     8    37%
      ICESTORM_HFOSC:     0     1     0%
         ICESTORM_LC:    47  5280     0%
      ICESTORM_LFOSC:     0     1     0%
        ICESTORM_PLL:     0     1     0%
        ICESTORM_RAM:     0    30     0%
      ICESTORM_SPRAM:     0     4     0%
              IO_I3C:     0     2     0%
               SB_GB:     1     8    12%
              SB_I2C:     0     2     0%
               SB_IO:     6    96     6%
         SB_LEDDA_IP:     0     1     0%
         SB_RGBA_DRV:     0     1     0%
              SB_SPI:     0     2     0%
         SB_WARMBOOT:     0     1     0%

CLOCKS:
                clki:   86.85 Mhz max

```

zapta, 2025-01-23:
@cavearr, the -dsp option seems to have some issues so it was a good call not to hardcode it in the yosys command.

https://github.com/YosysHQ/yosys/issues/4865

cavearr, 2025-01-23:
your apio.ini file had been a very great idea! thank @zapta !

zapta, 2025-02-04:
Fixed. Closing.

zapta, 2025-04-23:
The yosys dsp issue was resolved https://github.com/YosysHQ/yosys/issues/4865

On our side we just need to create a new version of the apio oss-cad-suite package with a newer yosys package that include it. This is independent of apio code and releases.