FPGAwars/apio #558: [For discussion] Program friendly reports from apio?

zapta, 2025-01-26:
@cavearr, this is a question for you.

Where does icestudio get the list of boards supported by apio?  

Will it be useful to add a program friendly output such as csv or json?  For example ``apio boards -csv``

cavearr, 2025-01-29:
Sorry @zapta! my mail server pranked me and sends some emails to the spam folder 😆 

Now the dashboards in Icestudio are hardcoded, this is something that could be interesting, but there are some problems, for example icestudio needs the pcf file and celery does not provide it, so this always needs to be hardcoded and packaged.

These types of commands could be interesting for the near future, in my preference they always output json but there is no rush if you have other priority tasks.

zapta, 2025-01-29:
We can have a dedicated command ``apio api ...`` with formatted output for
programs and keep backward compatible across apio releases.

zapta, 2025-01-29:
We can have a dedicated API command ``apio api ...`` with formatted output for
programs and keep it backward compatible across apio releases.

cavearr, 2025-01-29:
this could be pure gold @zapta! i'm doing something similar for icestudio.

zapta, 2025-02-05:
The first version is ready via the command ``apio api info -h``.  We will extend is as we go and based on needs. 

Closing.

closed 2025-02-05 by zapta