FPGAwars/apio #580: iCELink open fail!

div110, 2025-03-25:
Hello,
I'm trying to upload code to my iCESugar v1.5 with apio upload but it gives me following error:
```
$ apio upload
[Tue Mar 25 12:19:36 2025] Processing iCESugar_1_5
-------------------------------------------------------------------------------------------------------------------------------
icesprog hardware.bin
iCELink open fail!
scons: *** [upload] Error 255
================================================= [ ERROR ] Took 1.26 seconds =================================================

```
The board itself is not even listed when I do `lsusb` nor `dmesg`. 
Thing that makes me mad is that it worked once. I was even able to upload code to it to verify. Laptop went to hibernation and after that iCELink icon was gone and the device silent again just like before. Have the same issue across different computers.

I've enabled drivers in apio:
```
apio --ftdi-enable
apio --serial-enable
reboot
```
but nothing changed. Also tried installing every driver possible, still no mention (lsusb,dmesg)

I'm clueless, any ideas?
Thanks.

zapta, 2025-03-25:
Maybe the usb cable or board itself are damaged?

I am not familiar with that board, it seems to have an STM32 as the USB interface (vs the more common FTDI). Maybe @cavearr may help.

div110, 2025-03-25:
> Maybe the usb cable or board itself are damaged?
> 
> I am not familiar with that board, it seems to have an STM32 as the USB interface (vs the more common FTDI). Maybe [@cavearr](https://github.com/cavearr) may help.

> i'll try it this afternoon and tell you.

Turns out that for some reason the USB-C works only one way, meaning if i rotate it 180 degrees it won't work.
I feel betrayed. 

Thanks a lot for your time tho, really appreciate it.

closed 2025-03-25 by div110

zapta, 2025-03-25:
USB C has pairs of contacts, on opposing sides, and one of them can be bad.

I test USB cables with this tester https://www.aliexpress.us/item/3256807313556832.html

div110, 2025-03-26:
> USB C has pairs of contacts, on opposing sides, and one of them can be bad.
> 
> I test USB cables with this tester https://www.aliexpress.us/item/3256807313556832.html

ah, okay. Will try with different cable.