FPGAwars/apio #581: Release the next version of Apio

zapta, 2025-03-25:
Apio dev should be stable and complete now and ready for the testing and release process.  Release can be as a pypal/pip python package and/or the new installers.

@cavearr, @Obijuan, leaving this to you.

zapta, 2025-04-22:
Hi @cavearr, what are your plans for the next release of Apio?

The current dev version is feature complete and stable as far as I can tell.

cavearr, 2025-04-22:
Hi @zapta! sorry for my delay!, i'm very busy closing the development version of icestudio to release Apio and Icestudio at the same time.

I need a couple of weeks more or less, the next week i'm telling you to start to plan the release, ok?

And once again, thank you for all!

zapta, 2025-04-22:
Very good. Thanks Carlos, and let me know if I can help in any way.

zapta, 2025-06-21:
I marked a few issues here with the label `pre release`. Please verify that they are completed before you release.

zapta, 2025-07-08:
Hi @Obijuan, I have exhausted my TODO list and Apio should be ready to be released as 1.0.x.  

Notes:

1. You can start with a release of a pypal pip package and then add the standalone installers and packages from here https://github.com/FPGAwars/apio-dev-builds/releases

2. Preferably the pypi package and the installers should come from the same apio commit SHA. You can run the installers builder for a specific apio commit here https://github.com/FPGAwars/apio-dev-builds/actions/workflows/build-and-release.yaml

3. The Quick Start and installation docs are here https://fpgawars.github.io/apio/quick-start, please review that they work and that the initial user experience is good.

4. The apio 'make' script was replace with an 'invoke' script (tasks.py). It has the `invoke publish-test` and `invoke publish-prod` commands but they are not tested, you may need to tweak them.

cavearr, 2025-12-05:
Thanks @zapta , great job, the community will appreciate this new version! @Obijuan we'll be releasing it tomorrow, Saturday afternoon-night. I'll get everything ready so it's all set. If you have any problems, let us know.