FPGAwars/apio #584: Cannot upload to `SRAM`,  APIO upload always adds `-f`

milewski, 2025-04-05:
When running the following command:

```
apio upload --sram
```

It internally executes:

```
openFPGALoader _build\hardware.fs -b tangnano20k -f
```

However, the `-f` flag should not be included, as the `--sram` option was explicitly specified. 

The inclusion of `-f` forces a flash upload, which contradicts the intended SRAM-only behavior.

Expected Behavior:

```
apio upload --sram
```

should result in

```
openFPGALoader _build\hardware.fs -b tangnano20k --sram
```

Why the SRAM isn't the default? I am using the development branch by the way

zapta, 2025-04-30:
Can you share your apio.ini?

zapta, 2025-04-30:
It seems that the ``--sram`` and ``--flash`` are flags are limited to specific programmers. 

We should revisit them and consider:
1. An error messages if used outside of their supported scope.
2. Clarification in the ``apio update -h`` text.

Alternatively, generalize the concept of user provided flags to the programmer, e.g. having multiple programmer variants in the board definition (``default``, ``sram``, etc) or allowing the user to find tune the programmer command in ``apio.ini``.

@milewski, as a workaround, you can add to your project boards.jsonc and/or programmers.jsonc with the custom definitions you want.  you can find the original ones here https://github.com/FPGAwars/apio/tree/develop/apio/resources and you can include in your custom files, just the definitions you actually use.

zapta, 2025-04-30:
@Obijuan, @cavearr, a question for you. The ``--sram`` is enabled for the ``iceprog`` programmer. Does it mean that it's really available for all boards that use ``iceprog`` or just for a few boards that are implemented with sram?  And, same question for ``--flash`` and the ``ujprog`` programmer.

https://github.com/FPGAwars/apio/blob/4d9fc1f1632a0ad902c24115b0794b5763be2560/apio/managers/programmers.py#L194

cavearr, 2025-04-30:
Hi @zapta , not all the boards, support --sram but is very interesting and useful  clarify this option. In icestudio is not included but my plan is add it because in the boards that support it the development time increases a lot.

zapta, 2025-04-30:
Hi @cavearr, I think we need to abstract better the way we present the programming modes to the user.  For example, something along these lines:

* Every board has one or more **programming modes**
* Each programming mode has a name, unique within the board.
* For each board, one programming mode is defines as the default. This can be overriden in apio.ini.
* When the user runs ``apio upload`` apio will use the default programming mode unless if it is overriden by the user  in  with a flag such as ``--mode sram``
* If the user specifies a programming mode that is not available for the current board, apio exists with an error.
* The programming modes of a board can be using the same or different programmer binaries.
* The command ``apio boards`` will list the programmers and the modes available for each board.
* The command ``apio api info --section boards`` will inform icestudio what modes are available for each board.
* We will use programming naming conventions such as ``sram`` what board board definitions can have unique names that are specified for their design and requirements.

Any thoughts?

cavearr, 2025-05-03:
Sorry for the delay, @zapta. I'm involved in a very important change to the Icestudio core, and it's taking me several weeks, longer than I expected.

I'll try, give me a moment.

cavearr, 2025-05-03:
Hi @zapta , i'm testing and with the same result as you, in my case the leds appears with a little light, but clearly is wrong.

I'm researching now, telling you soon.

zapta, 2025-05-03:
When running:

 apio raw -- iceprog --help

It gives at the end of the text some information about board modification
for SRAM mode.

cavearr, 2025-05-03:
Hi @zapta ! Bad luck. Apparently, it's not possible. The Alhambra-II doesn't have the pins configured to support sram recording.

I'll try it tomorrow on the rest of my boards with Ice40 to see if this mode works on any of them.

Thanks for the great work!

cavearr, 2025-05-03:
Hi @zapta ! yes, i review this documentation before my latest post and review the alhambra-ii board schematics, and the problem is that this pins are not possible to reconfigure without "cut" paths in the alhambra PCB XD

zapta, 2025-05-03:
@cavearr how about removing the --sram option all together?  

It is a confusing special case with too many restrictions that has a simple workaround using the command ``apio raw -- iceprog ...``.

cavearr, 2025-05-04:
Hi @zapta ! Since it's done, I'd leave it. In the near future, it might be useful for new boards. I'm not aware of all the integrated boards (I don't have them all, and since I only had Flash programming at the beginning, I haven't really tried this feature on all of them). My plan was to gradually test all the boards throughout the week (although I don't have them all, I have quite a few) to see which ones support the feature and which ones don't. I think Colorlight, for example, all support both modes.

It's a feature I was really looking forward to implementing, and I wasn't aware of the limitation on the Alhambra, but as I said, since you've implemented it, I think it's worth leaving it.

zapta, 2025-05-04:
@cavearr, sounds good. I sent you a pull request with help text that explains the --sram flag restrictions. 

https://github.com/FPGAwars/apio/pull/588

zapta, 2025-05-04:
The limitations of ``iceprog`` SRAM mode are now reflected in error messages and the help text of ``apio upload``.  Closing this one.

Closing this one.

cavearr, 2025-05-10:
@zapta, I made an incredible discovery. I'm integrating fpgaloader for some IceStudio plugins and, reading the documentation, I found it had JTAG and Dirty JTAG support for FTDI chips. Although the Alhambra-II isn't strongly integrated (for example, it doesn't detect it), it can be used as a generic Ice40-based card.

So, I quickly tried saving to SRAM as if it were a generic Ice40...and it worked! Just type the command if you want to try it:

````
openFPGALoader -b ice40_generic _build/hardware.bin
````

I'm attaching a screenshot:

[image]

Here's the list of boards supported by OpenFPGALoader (it's already in the Yosys package),
to see ice40_generic-type identifiers so we can use it on boards that run IceProg and have the same problem:

https://trabucayre.github.io/openFPGALoader/compatibility/board.html

I'm reopening the issue in case you want to address it. I think it's a super important feature now that we know it can be easily done.

zapta, 2025-05-10:
Hi @cavearr, do you suggest to switch the alhambra-ii board and possibly other boards from the iceprog programmer to the openfpgaloader programmer?  @Obijuan mentioned in the past something to that effect.

https://github.com/FPGAwars/apio/issues/536#issuecomment-2586341714

cavearr, 2025-05-10:
@zapta, @obijuan , could we use openfpgaloader  only for sram programming? i vote for the full switch to openfpgaloader (i'm using a lot for other things and explore in deep for some plugins in icestudio) and i think if we tested well ,should not be problematic.

For the users should be totally transparent and for the moment we could do for alhambra-ii to test until migrate other boards.

I think the sram programming is a very big improvement.

zapta, 2025-05-11:
@cavearr, I created an issue for migrating boards to openfpgaloader, please respond there.  https://github.com/FPGAwars/apio/issues/596

Closing this one since we documented at 'apio upload -h' the limitations of the --ram option (applies to only some boards and may require board changes).

closed 2025-05-11 by zapta