FPGAwars/apio #592: Add support for VHDL

zapta, 2025-05-07:
This is the canonical request for VHDL support.

See duplicates that are linked here for details.

cavearr, 2026-09-25:
@zapta, I’m working on integrating metalanguages ​​into Icestudio, but I think Apio is the right place for that. What do you think? If you don't think it makes sense, I’ll continue with the work there.

I’m referring to metalanguages ​​like Silice or Amaranth, for example; I think they could be handled in the same way as VHDL and Verilog.

Would you be okay with me drawing up a plan for this and taking charge of it? I think it would be a very interesting feature for the coming months.

zapta, 2026-09-26:
Hi @cavearr, it would be useful to add support for additional languages as long as they are reasonably mature and have sufficient demand. Most likely VHDL will have demand but I don't know how well it's supported by yosys and if needed by all the architectures we currently support.

Can you experiment with a few VHDL examples (without Apio) to get the feel of it?  Currently we allow to mix in the same project .v (verilog) and .sv (system verilog) and it would be nice if we cal also mix .vhdl.

zapta, 2026-09-26:
Hi @cavearr, this is Grok's estimation of language usage in open source FPGA projects.

[image]

zapta, 2026-09-26:
According to Grok, these are the languages that are supported by oss-cad-suite

[image]

cavearr, 2026-09-26:
Yes, I have that table in mind, but I don't think it's possible to mix VHDL and Verilog directly; that would be a different story if we were trying to create an unconventional architecture.

I think it would be better to experiment in Icestudio using direct Apio commands; once it's working and we have user feedback, we could consider how to delegate that responsibility to Apio. What do you think? We could keep this issue open while I experiment in Icestudio.

zapta, 2026-09-26:
Hi @cavearr, when you say "I think it would be better to experiment in Icestudio", what language do you have in mind, VHDL or some metalanguage?

My thinking is that since icestudio is schematic based, adding a new  text based language is less important than working on a great user experience out of the box. I didn't play with icestudion in the last 1-2 years because I had hard time to find a release for macos but from what I remember there were some aspect of the user experience that could be streamlined, for example, having a clear notion of a 'project' that can be opened, closed, etc, or make it easier to find existing libraries and components.  That was 1-2 years ago so maybe it's better now but if you want, I can play with it again and give you feedback.

cavearr, 2026-09-26:
Hi @zapta, I'm making huge progress with Icestudio, I already released a version with some minor improvements this summer, and the community received them very positively, but that is just the tip of the iceberg.

Wait until I let you know in a couple of weeks; by then, I’ll have uploaded the version I want to finalize as 1.0, which will include a simulation engine and major IDE improvements.

For me, Icestudio is going to be more than just a block-based environment; the idea is for it to serve as a full-fledged IDE with professional-grade capabilities (simulation, IDE features, ILA, blocks, and more).

If everything goes well, it’s going to be a huge leap forward.

The macOS issue is already resolved; it installs just as easily as any other Mac app, with no problems at all.

However—regarding the effort of testing it and providing feedback, which I’d ​​really appreciate—please wait until I give you the word; it’ll be much more productive that way.

And as always, a million thanks.

cavearr, 2026-09-26:
@zapta ...and regarding metalanguages ​​and VHDL. I was evaluating VHDL, but also the others (Silice and Amaranth).

IceStudio makes it easy for me to experiment with these languages. As I mentioned earlier, the idea behind IceStudio is to function as an IDE, a block manager, and a connector between blocks, while also simplifying the handling of boards and other tools—such as simulation, ILA, etc.—using your preferred editor. For instance, I’m currently rolling out integration with external files so you can work in parallel—using Visual Studio for code and IceStudio for the design—with automatic synchronization between them.

This is crucial because, at the block level, IceStudio will never be anything more than a "toy" tool; however, with the right tools and an expansion into a text-based IDE, it could become the first real professional competitor to proprietary IDEs—and that is my goal.

zapta, 2026-09-26:
Hi @zapta, that sounds very good. Ideally icestudio will allow to mix 'blocks' that was implemented in different ways, verilog, system verilog, schematic, synthesis, tests, interactive simulation, internal and external editors, and in the future other languages.

At this point, I think that the effort to add new languages may distract from the mission of making the existing features flexible and easy to use, there is plenty of work to do without it.

cavearr, 2026-09-26:
yes, for me this is a task after the icestudio 1.0, one of the things i'll do then, will be regarding showing and planning an organized roadmap—that was really just a "Hey Zapta! I’m looking into this too, and if you're interested, I can put together a proposal for you once I've got something ready."

I’m going through the issues to see if there’s anything else I can help with, and I keep coming across things that weren't previously on my radar; my main goal is to ensure we're aligned and don't waste any effort.

It’s a real pleasure working with you—things flow really well.