FPGAwars/apio #596: Migrate some boards to openfpgaloader programmer

zapta, 2025-05-11:
The openfpgaloader programmer is already used by some boards and migrating additional boards to it where surfaced in a few discussions:

https://github.com/FPGAwars/apio/issues/536#issuecomment-2586341714
https://github.com/FPGAwars/apio/issues/584#issuecomment-2869123006
https://github.com/FPGAwars/apio/issues/584#issuecomment-2869139184

@Obijuan, @cavearr, please respond here with this info

1. What is the motivation for migrating boards to openfpgaloader? What do we expect to achive?
2. Which boards are candidate for this migration?
3. Which of the candidate boards we can actually test?  (that is, we have access to those boards)

cavearr, 2025-05-11:
Hi @zapta !

1) The motivation is do support for direct fpga programming (without flashing the SPI) in som FTDI+SPI+ice40 (like icestick or alhambra) boards and that iceprog can't do it.

2) for the momento i think we could only migrate alhambra-ii and icestick to test the new setup

2) Alhambra-ii

Other strategy could be from icestudio use the apio raw comand to implement  the on the fly programming and when the users do feedback translate and apply in Apio.

If you prefer this option we could do it in this direction, and for the moment do nothing in apio.

zapta, 2025-05-11:
Thanks @cavearr, if it's just alhambra-ii it should be simple to try since I have the board. I will try it from 'apio raw' command line and go from there. alhambra-ii is the apio reference design so will be good to have the --sram works out of the box.

zapta, 2025-05-12:
Hi @cavearr, I was able to upload to Alhambra-ii FLASH using the command below. 

```
$ apio raw -- openFPGALoader --verify -b ice40_generic _build/default/hardware.bin
```

What command do you use for SRAM?  When I add the ``--write-sram`` flag it hangs on ``Wait for CDONE`` and then fails.

```
$ apio raw -- openFPGALoader --write-sram --skip-reset  -b ice40_generic _build/default/hardware.bin
empty
write to ram
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file DONE
Loading to CRAM: [==================================================] 100.00%
Done
Wait for CDONE FAIL
```

cavearr, 2025-05-15:
Hi @zapta ! to program directly into the FPGA sram you only need to use:

openFPGALoader -b ice40_generic bitstream.bit

zapta, 2025-05-15:
Ok, I missed your other post ;-)  https://github.com/FPGAwars/Alhambra-II-FPGA/issues/13#issuecomment-2882772588

zapta, 2025-05-15:
@cavearr, when I try the command below the programming seems to go to FLASH  (I reconnect the power to the FPGA board and check if the program is still there). Am I missing something?

```
~/work$ apio raw -- openFPGALoader -b ice40_generic _build/fast/hardware.bin 
empty
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file DONE
JEDEC ID: 0xef4016
Detected: Winbond W25Q32 64 sectors size: 32Mb
00
JEDEC ID: 0xef4016
Detected: Winbond W25Q32 64 sectors size: 32Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00030000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Wait for CDONE DONE
```

cavearr, 2025-05-15:
Sorry, @zapta. OpenfPGAloader is very fast compared to iceprog. I got carried away; I should have checked more thoroughly.

In any case, the change could be very positive; as I said, it's true that it's very, very fast.

On the other hand, I'm running some tests because I think it's possible to save to Flash with a custom programmer. I'm exploring possibilities. If I can get it working, I'll let you know, and we'll see what we can do.

Thank you very much for everything.

zapta, 2025-05-15:
@cavearr, openFPGALoader seems to be a good choice (feature rich, fast, and actively maintained) and I plan to transition to it also internal apio functions that currently use 'lsftdi'  (I created an apio issue about it).

There is one more thing that we need to test. Alhanbra-ii uses a 4K FPGA that we use as a 8K FPGA. Can you confirm that this trick also works with openFPGALoader?  That is, have a larger than 4K design and see if it works using openFPGALoader programming.

zapta, 2025-06-21:
Hi @cavearr, did you have a chance to test the 7060 LCs example?  I want to make sure that the Alhambra-ii 8K claim is still valid with openFPGAloader.

zapta, 2025-06-21:
Created https://github.com/FPGAwars/apio/issues/644 for the 8K verification.

Closing this one.

closed 2025-06-21 by zapta