FPGAwars/apio #597: Remove the temporary support for legacy board names.

zapta, 2025-05-11:
At this time, apio dev contains logic to support old board names.  That include replacing instances of old names with the new names and printing a warning message to the user asking them to fix apio.ini.

``legacy-name`` field in boards.jsonc   https://github.com/FPGAwars/apio/blob/a7aedd86be72201b9dab809eb59ac6ca295a1d3e/apio/resources/boards.jsonc#L71

Logic to replace old board name in apio.ini "board" options to their respective new name  https://github.com/FPGAwars/apio/blob/a7aedd86be72201b9dab809eb59ac6ca295a1d3e/apio/managers/project.py#L210

This issue is to delete this backward support after the next release.

cavearr, 2026-09-16:
@zapta  Looks already done: there is no `legacy-name` left anywhere under `apio/` on `main`, and `project.py` no longer maps old board names (the only legacy handling left there is the `[env]` section, which is #598). 

The apio-vscode 0.1.9 changelog records the removal ("Removed backward compatibility with old board names"). Proposing to close.

zapta, 2026-10-16:
Legacy name field removed. Closing.

closed 2026-10-16 by zapta