FPGAwars/apio #607: Do we need 'apio drivers install ftdi' and _pre_upload_darwin()

zapta, 2025-05-23:
While working on the apio upload code I encountered old darwin specific logic that doesn't seem to do anything on my system, and it's possibly for pre osx 11 which was released about 5 years ago.  This issue is to figure out if we still need this extra complexity.

1. Do we need the ``apio drivers install ftdi`` which runs ``brew install libffi`` and ``brew install libftdi``.  I can program alhambra-ii on my system without these brew modules.

2. Do we need the pre/post programming hooks that are triggered if ``libftdi`` is installed and if ``com.FTDI.driver.FTDIUSBSerialDriver`` or ``com.apple.driver.AppleUSBFTDI``. On my system I have neither of them and everything works.  The code (about 8 years old) is here:

https://github.com/FPGAwars/apio/blob/e2bb0f46f03dddbe580e7f0a250b55746897500e/apio/managers/drivers.py#L435

It's possible that this is a legacy leftover code that just adds unnecessary complexity.

@cavearr, @Obijuan, any thoughts or historical context?

zapta, 2025-05-23:
@cavearr, I think that the libftdi comes with yosys' oss-cad-suite and if I delete it openFPGALoader fails to upload.  This suggests that installing brew libftdi is unnecessary.  I will try to see what yosys says about their installation requirements.

```
~/.apio/packages$ rm ./oss-cad-suite/lib/libftdi1.2.5.0.dylib

~/.apio/packages$ ww
~/work$ apio upload
Using env slow (alhambra-ii)
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Setting shell vars.
Querying programmer parameters.
Error: The command 'openFPGALoader --scan-usb' failed with error code -6
```

cavearr, 2025-05-23:
Good job @zapta! i'm confirm in clean osx you don't need install drivers ;) We could remove this step in osx.

In Windows, we need to try well  if zadig is needed or not. In windows 10 (we have users and schools, with this legacy system)  is needed but i think in windows 11 could be not  needed, i'll try it in the weekend.

zapta, 2025-05-23:
Thanks @cavearr, this is a great help. I am trying to simplify the path of the 'apio command'.

zapta, 2025-05-23:
Hi @cavearr, I presume that we also don't need the darwin serial drivers. The two libraries ``libusb`` and ``libffi`` are already in the ``oss-cad-suite`` package.

Does it make sense?

https://github.com/FPGAwars/apio/blob/e2bb0f46f03dddbe580e7f0a250b55746897500e/apio/managers/drivers.py#L405