FPGAwars/apio #613: Restore the tinyprog special logic

zapta, 2025-05-29:
The ``apio upload`` logic is being rewritten. This issue is to restore the special logic of the tinyprog programmer.

Search for 'tinyprog' in these files:

https://github.com/zapta/apio-dev/blob/3dd704b4173391e2c52ad316dbba0f7810931730/apio/managers/programmers.py#L3

https://github.com/zapta/apio-dev/blob/3dd704b4173391e2c52ad316dbba0f7810931730/apio/utils/util.py#L1

Encapsulate the tinyprog specific logic in a new ``tinyprog_util.py`` file and call it as needed.

zapta, 2025-06-17:
Hi @cavearr, do you have a board for testing the tinyprog programmer?  Any idea where I can get one?

cavearr, 2025-06-17:
yes i have the tinyfpgabx but it have serial problems with the latests tinyprogs, i did a mod few months ago for a user request that launch the alert. I need to packetize and prepare to replace the "oficial one" that is outdated. The project appears abandoned, i think in mouser are selling it but in the oficial project there are no maintainers replys for several years.

This is an interesting board for some concepts but i think this is not the best  board to buy now. If you want to buy a very interesting board, gets a ulx3s or colorlight i9 , for me this boards are very useful and powerful.

cavearr, 2025-06-17:
@zapta Even though the project is obsolete, there are tons of people using the tiny, and there are also several posts on hackster.io talking about celery and the tiny, so I wouldn't remove it.

If you'd like, we can do two things. On the one hand, in the next few days I can restart the programmer to finalize the version, and I'll post a post so users who are having problems can try it and give us feedback. The problem only occurs on cards from the last 2024 releases (I think starting in May 2024). The older ones work fine.

And on the other hand, if you want, I can connect the tiny to a computer and give you SSH access to an environment, and with a webcam or something like that, I can see how to set up some kind of remote lab so you can play with the tiny and run tests. I wouldn't be able to do it until the weekend, but if you leave it on standby, I think I could have it set up by Sunday or Monday at the latest. If it motivates you, let me know and write me directly to my email so we can talk privately (if you're on Discord, my username is cavearr).

If this last option doesn't motivate you, don't worry; it's something I came up with on the fly.

zapta, 2025-06-18:
Hi @cavearr, remote debugging is an interesting idea. Let me try first with unit tests and debug messages to see how it goes.

I looked more carefully into the tinyprog issue and it seems that it should work now, if there is only one tinyprog device connected to the computer.  Can you put this in your `apio.ini` and see if it works?

```
programmer-cmd = tinyprog --pyserial  --program
```

If it will work we can move that definition to the link below and avoid the special logic: https://github.com/FPGAwars/apio/blob/ac0073b45a63137c19bdc52a8ec0ab350f96776b/apio/resources/programmers.jsonc#L59

zapta, 2025-07-05:
Per https://github.com/FPGAwars/apio/issues/642 tinyprog works well as is. Closing this one.

closed 2025-07-05 by zapta