FPGAwars/apio #641: Delete the old apio doc (wiki)

zapta, 2025-06-21:
This issue is for deleting the old Apio documentation to avoid confusion with the new documentation and command changes.  After the deletion, the Wiki can be a good place for less formal and not closely maintained additional Apio related information.  If there is anything missing in the new docs, add it.

Old doc : https://github.com/FPGAwars/apio/wiki

New doc: https://fpgawars.github.io/apio/

cavearr, 2025-06-28:
@zapta @Obijuan ,  It took me a while, but I finally figured out how to redirect the old documentation to the new one.

Let me know if you have any comments; I'll consider this closed. Any improvements or comments, please reopen the issue.

cavearr, 2025-06-28:
Sorry @zapta ! i don't understand well the title 🤒

zapta, 2025-12-31:
Done. Closing.