FPGAwars/apio #642: Test 'apio upload' with tinyprog board

zapta, 2025-06-21:
We have made several changes to the post processing of the scons output which may break the progress bar of the tinyprog programmer. I was don't have a compatible board and was not able to test it.

@cavearr, please the 'apio upload' command of the latest apio with a tinyprog board.

Thinks go look for.
* The output during the upload looks reasonable and the progress bars operate corectly.

You can compare it with running the same programmer command with the `apio raw -- ` prefix.

cavearr, 2025-07-01:
This works well with old firmware. i'll try to get a new device to test and provide a patch for our tools but for the moment i'm closing this issue, doing it for done.

closed 2025-07-01 by cavearr

zapta, 2025-07-01:
Thanks @cavearr!!