FPGAwars/apio #644: Confirm that Alhambra-ii has 8K with openFPGAloader

zapta, 2025-06-21:
We migrated Alhambra-ii to use openFPGAloader instead of iceprog. This issue is to confirm that it still support 8K designs (despite the fact that it has a nominally 4K fpga).

@cavearr, can you confirm that it works?

zapta, 2025-06-21:
Hi @cavearr, a few Apio issues are assigned to you. Please find the time to take a look. You can find them here https://github.com/FPGAwars/apio/issues?q=is%3Aissue%20state%3Aopen%20assignee%3Acavearr

cavearr, 2025-06-22:
Hi @zapta , i have a ton of work with icestudio this months, sorry for my delay, between today and tomorrow, i'm check and review all, count with it.

cavearr, 2025-06-29:
Hi @zapta , all works fine, i did an example that fills in 90% of LCs the FPGA and use all  BRAMS (32 for 8k fpga), all works fine in Alhambra-ii board (ice40hx4k treat as ice40hx8k):

[image]

I'm push this new example to apio as reference in some minutes.

cavearr, 2025-06-29:
sorry the report has an error, this is the output for the last area test design:

<img width="373" alt="Image" src="https://github.com/user-attachments/assets/3a714c94-fa8b-47e3-8cdf-6282b216ff14" 

I'm pushing to alhambra-ii examples now.

zapta, 2025-06-29:
Thanks @cavearr!