FPGAwars/apio #647: Test apio colors on Windows 11

zapta, 2025-06-23:
This issue is to test the apio colored text on windows 11 machine (which I don't have).

* Test with the standard windows shells (CMD, Power shell, etc)
* If possible test with git bash shell 

Run the following commands in an apio project and verify that the text coloring is correct and that you don't see raw ANSI escape sequences.

* `apio info colors`  - this should display a page with colored text
* `apio format` - formatted file names should be colored
* `apio report` - some cells in the table should have colored text, table frame lines should be solid with no breaks.
* `apio build` (run twice) - The `scons: build is up to date` message should be colored.

Then run `apio preferences --theme no-colors`, run the text again and see that everything looks good and with no color whatsoever.

@cavearr, assigning to you.  ;-)

cavearr, 2025-06-28:
@zapta , i'm confirming you that all the above commands, shows very well on windows 11 , in **cmd** and **powershell**

Attaching info colors output in both shells:

[image]

closed 2025-06-28 by cavearr

zapta, 2025-06-28:
Very good. Thanks @cavearr!